realsenseai/librealsense #7175: L515 RGB streaming error in RealSense Viewer after switching on Motion Module.

thupalo, 2020-08-24:
| | |
|---|---|
|**librealsense**|2.36.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense L515|
|**Serial Number**|00000000f0220921|
|**Firmware Version**|01.04.01.02|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|L500|
|**Asic Serial Number**|00000003ac6c5af7|
|**Firmware Update Id**|00000003ac6c5af7|

Please provide a description of the problem:
++++++++++++++++++++++++++++
After switching on Motion Module, the RGB stream stops with flashing info "No Frames Received!".
Nothing helps but Viewer restart.

console message:

 20:53:38 [Warn] C:\jenkins_sys_rsbuild\workspace\LRS_windows_compile_pipeline\src\rs.cpp:1089 - null pointer passed for argument "frame"

RealSenseSupport, 2020-08-24:
Hi,

This is most likely due to a known OpenVino Issue. Try disabling OpenVino Face Detection in the RGB Camera Post Processing Settings in RealSense-Viewer to see if that solves the problem.

Thanks

thupalo, 2020-08-24:
Hi,
After switching off Face detection - it works!
Thank you for fast response, 
Stay safe

RealSenseSupport, 2020-08-24:
Glad that worked.

You as well!

closed 2020-08-24 by RealSenseSupport