realsenseai/librealsense #7176: exttrinsics_calibrate for L515 won't work

AndreV84, 2020-08-24:
ref: https://github.com/IntelRealSense/librealsense/commit/acb0bbd0efb10ecf81343209d83aeb6cb30e23a2
ref: https://github.com/schmidtp1/librealsense/blob/calibrate-extrinsics/wrappers/python/examples/calibrate_extrinsics.py
The question is how to pass L515 serial number to the ccode so that it works?
Is it possible?

dorodnic, 2020-08-31:
Looking at the script, there are two things different for L500:

1. There is no IR2
2. IR1 in L500 is called IR0

AndreV84, 2020-08-31:
Thank you for the update!

AndreV84, 2020-08-31:
@schmidtp1 
is there a variant for l515?
@dorodnic do you suggest justy removing lines containing IR2 & renaming IR1 to IR0? In order to make it work for L515?

schmidtp1, 2020-08-31:
Hi @AndreV84, yes, I think you should be able to replace the second image stream, e.g. by IR0 of the L515. The script reads the intrinsic calibration (using LRS API) so that should fit.

AndreV84, 2020-08-31:
@schmidtp1 Thank you for your response! So do I rename ,in all lines mentioning IR2, the latter value to  theIR0, then try with l515 serial? Or also IR1 lines will need to be modified?

AndreV84, 2020-08-31:
```
 python3 calibrate_extrinsics.py --SN_T265 11000000000 --SN_D4xx 0000000000000000
Trying to connect devices...
Traceback (most recent call last):
  File "calibrate_extrinsics.py", line 103, in <module>
    sensor_depth.set_option(rs.option.emitter_enabled, 0)  # turn OFF projector
RuntimeError: object doesn't support option #18
```
the result above is after editing all ir2 -> ir0
@schmidtp1 would you be able to help getting through the adjustmewnt of the code?

AndreV84, 2020-09-02:
whhere to get calibration chessboarrds for the caibration?

schmidtp1, 2020-09-15:
Hi @AndreV84, sorry for the late reply. The script might require some minor modifications since it was never tested with L515. You can try to just comment the lines related to disabling the projector.

Regarding the calibration target, it is a standard chessboard. The important part are the dimensions:
```
parser.add_argument('--grid_H', default=9, help='grid height (inner corners)')
parser.add_argument('--grid_W', default=6, help='grid width (inner corners)')
parser.add_argument('--size', default=0.020, help='grid side length')
```
(https://github.com/schmidtp1/librealsense/blob/calibrate-extrinsics/wrappers/python/examples/calibrate_extrinsics.py#L23 ff.)
This link might be helpful: https://docs.opencv.org/master/da/d0d/tutorial_camera_calibration_pattern.html.

AndreV84, 2020-09-15:
Thank you for following up!

RealSenseSupport, 2020-09-30:
Is there anything else needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

AndreV84, 2020-09-30:
why doesn't the www.intelrealsense.com/dimensional-weight-software/ activate with L515?

RealSenseSupport, 2020-10-07:
Hi @AndreV84 

Can we close out this thread?  The DIM Weight SW activation item has been handled via other support process already.

*If we don’t hear from you in 7 days, this issue will be closed.

AndreV84, 2020-10-07:
done

closed 2020-10-07 by AndreV84