realsenseai/librealsense #7177: DWS sample code

rvermeire, 2020-08-24:
Hi, 

I am running the DWS sample code referenced from Intel_RealSense_DWS_BETA_v1.pdf and online here https://www.intelrealsense.com/download/13022/ . The code is not identical and in both cases not working, because of lots of signature changes i guess. Could you provide a working example or should I post my code so we can check the errors one by one? SDK version is _v0.7.5_RC12

Kind regards,

Ronald Vermeire

RealSenseSupport, 2020-08-28:
Hi @rvermeire 

We are looking into this.  Will let you know when we have something.

There is also sample code in the documentation that is in the readme folder of the zip download package:  Intel_RealSense_DWS_BETA_v1.pdf

rvermeire, 2020-08-28:
Hi,

Thanks for looking into this.

Yes I know there is code both online and inside the PDF and that's what I mention at the start of the issue:

> I am running the DWS sample code referenced from Intel_RealSense_DWS_BETA_v1.pdf and online here https://www.intelrealsense.com/download/13022/ . The code is not identical and in both cases not working

Kind regards,

Ronald Vermeire

yschleif, 2020-08-31:
Hi,

Regarding the lib file, it was left out of the release by mistake, and will be fixed. in the meantime, you can create it by following direction [here](https://wiki.videolan.org/GenerateLibFromDll/).

As for the sample code, obviously there was a problem.. Following is the updated code, which has some more detailed comments:

````
#include "DWS_sdk.h"
#include <iostream>
#include <mutex>

void * FrameBuffers[3]; // 0 - depth, 1 - IR, 2 - RGB
void * local_depth = new unsigned char[640 * 480 * 2];
void * local_ir = new unsigned char[640 * 480 * 1];
void * local_colour = new unsigned char[1280 * 720 * 3];
bool keep_alive = true;
bool image_ready = false;
std::mutex mutex_display;
DWS_Box boxes[DWS_MAX_NUM_OF_BOXES];
int numOfDetectedBoxes = 0;
long im_id = -1;
int handler = -1;

void * _callback(const void* depth_buffer, int d_w, int d_h, int d_bpp, const void* ir_buffer, int i_w, int i_h, int i_bpp, const void* colour_buffer, int c_w, int c_h, int c_bpp, int frameId)
{
	// ###############################
	// #### CRITICAL ####
	// COPY FRAMES TO LOCAL BUFFER!!!!
	// ###############################
	{
		std::lock_guard<std::mutex> lck(mutex_display);
		memcpy(local_depth, depth_buffer, (d_w *d_h *d_bpp));
		memcpy(local_ir, ir_buffer, (i_w *i_h *i_bpp));
		if (colour_buffer!=nullptr)
			memcpy(local_colour, colour_buffer, (c_w *c_h *c_bpp));
		im_id = frameId;
		image_ready = true;
	}

	return nullptr;
}

void * _dim_callback(int frameId, int box_count, const DWS_Box* box_buffer)
{
	numOfDetectedBoxes = box_count;
	return nullptr;
}

void* _status_callback(DWS_Severity severity, DWS_MessageCode code, const char* msg)
{ 
	std::cout << "Severity: " << severity << ", Code: "<< code << ", " << msg << std::endl;

	switch (code) {
	case CALIBRATION_FAILED:	// Couldn't complete base surface detection
	case BAD_USER_INPUT:		// Bad input to dws_create
	case CAMERA_DISCONNECTED:	// Detected camera disconnection
	case INVALID_LICENSE:		// License file missing, expired, or doesn't match camera SN
	case MISSING_CONFIGS:		// Couldn't find the /Configuration/ folder
	case BAD_API_CALL:			// API handle called during an active earlier one
	case UNKNOWN_ERROR:			// Internal, exception, msg might contain more details
		dws_stop(handler);
		keep_alive = false;
	default:
		break;
	}

	if (code == CALIBRATION_SUCCESSFUL)
	{
		if (dws_start(handler, _dim_callback) != DWS_Status::SUCCESS)
			dws_stop(handler);
	}
	return nullptr;
}

int main(int argc, char* argv[])
{	
	int imageNum = 0;
	auto status = dws_create(DWS_Mode::STATIC_MODE, false, _callback, _status_callback, &boxes[0], handler);
	if (status != DWS_Status::SUCCESS)
	{
		return -1;
	}
	std::string _ver = dws_version(handler);
	dws_autoCalibrate(handler, false);
	while (keep_alive) {
		if (image_ready) {
			std::lock_guard<std::mutex> lck(mutex_display);
			std::cout << "new image received. ID = " << im_id << std::endl;
			image_ready = false;
		}
	}

	return EXIT_SUCCESS;
}
````

rvermeire, 2020-08-31:
Hi

Thanks for the updated sample. Now method dws_autoCalibrate is not known. I used dws_autoInit instead, but it fails with: 

>  No floor calibration data detected. Creating new initialization flow.

BAD_USER_INPUT and BAD_API_CALL constants are also not known in my version of the SDK.

I have the viewer running in parallel and it is able to recognize the callibration target and measure a box after that.

Full console log:

> [2020-08-31 17:36:52.331] [DWSSDK] [info] [DWSSDK] LOG CREATED
[2020-08-31 17:36:52.332] [DWSSDK] [info] [DWSSDK] SDK created with mode=0
[2020-08-31 17:36:52.333] [DWSSDK] [info] [DWSSDK] OMP threads: 8
[2020-08-31 17:36:52.334] [DWSSDK] [info] [DWSSDK] SDK_create
[2020-08-31 17:36:52.334] [DWSSDK] [info] [DWSSDK] looking for files in: C:\*\static_mode_config.json
[2020-08-31 17:36:52.336] [DWSSDK] [info] [DWSSDK] Successfully parsed configuration
[2020-08-31 17:36:52.741] [DWSSDK] [info] [DWSSDK] Preparing background worker...
[2020-08-31 17:36:52.742] [DWSSDK] [info] [DWSSDK] Initializing player...
[2020-08-31 17:36:54.872] [DWSSDK] [info] [STREAM] Prepare LibRealsense terminal...
[2020-08-31 17:36:54.903] [DWSSDK] [info] [STREAM] Prepare LibRealsense terminal... DONE
[2020-08-31 17:36:58.436] [DWSSDK] [error] [STREAM]
Failed to resolve the request:
        Format: Z16, width: 640, height: 480
        Format: Y8, width: 640, height: 480
Into:
        Formats:
         Y8
         Z16
[2020-08-31 17:36:58.443] [DWSSDK] [info] [DWSSDK] Initializing player... DONE
[2020-08-31 17:36:58.454] [DWSSDK] [info] [DWSSDK] LOG CREATED
[2020-08-31 17:36:58.455] [DWSSDK] [info] [STREAM] ======================
[2020-08-31 17:36:58.455] [DWSSDK] [info] [STREAM] Attempt to tun script: C:\*\logState.txt
[2020-08-31 17:36:58.457] [DWSSDK] [info] [STREAM] > irb e2 09 1  // laser modulation limit
[2020-08-31 17:36:58.500] [DWSSDK] [info] [STREAM] Offset: 0x0 => 2D
[2020-08-31 17:36:58.501] [DWSSDK] [info] [STREAM] > irb e2 0a 1  // laser modulation limit
[2020-08-31 17:36:58.548] [DWSSDK] [info] [STREAM] Offset: 0x0 => 2D
[2020-08-31 17:36:58.549] [DWSSDK] [info] [STREAM] > mrd a00e1890 a00e1894 // invalidation configuration
[2020-08-31 17:36:58.644] [DWSSDK] [info] [STREAM] Offset: 0x0 => 00000000
[2020-08-31 17:36:58.646] [DWSSDK] [info] [STREAM] > mrd a00e1600 a00e1604 // edges configuration
[2020-08-31 17:36:58.745] [DWSSDK] [info] [STREAM] Offset: 0x0 => 00000100
[2020-08-31 17:36:58.746] [DWSSDK] [info] [STREAM] > mrd a00e1894 a00e1898 // Confidence configuration
[2020-08-31 17:36:58.790] [DWSSDK] [info] [STREAM] Offset: 0x0 => 00000003
[2020-08-31 17:36:58.791] [DWSSDK] [info] [STREAM] > mrd a00e0b0c a00e0b10 // Sharpness configuration
[2020-08-31 17:36:58.832] [DWSSDK] [info] [STREAM] Offset: 0x0 => 0000000C
[2020-08-31 17:36:58.833] [DWSSDK] [info] [STREAM] > mrd a0020ca8 a0020cac // Rast configuration
[2020-08-31 17:36:58.877] [DWSSDK] [info] [STREAM] Offset: 0x0 => 00000000
[2020-08-31 17:36:58.878] [DWSSDK] [info] [STREAM] > apd_flyback_values_get // Get flyback values
[2020-08-31 17:36:58.920] [DWSSDK] [info] [STREAM] state:          5
isDataValid:    1
reserved16:     BE09
vSenseReq:      B701
pwmValue:       5D04
tSenseRead:     B701
vSenseRead:     B501
tSenseC:        45.85949707
vSeneseReadV:   -155.9743958
reqV:           -155.6328735
[2020-08-31 17:36:58.921] [DWSSDK] [info] [STREAM] > RdCalibInfo 12
[2020-08-31 17:36:58.922] [DWSSDK] [error] [STREAM] Command not found!
[2020-08-31 17:36:58.923] [DWSSDK] [info] [STREAM] ======================
[2020-08-31 17:36:59.027] [DWSSDK] [info] [BM_API] No floor calibration data detected. Creating new initialization flow.
C:\*\rs_dws_01.exe (process 73132) exited with code 3.
Press any key to close this window . . .

Kind regards,

Ronald

yschleif, 2020-08-31:
Sorry, got ahead of myself and gave you our newest code that has some API changes from the last release.
Regarding your log:
"No floor calibration data detected. Creating new initialization flow." is not an error, it's just that the SDK hasn't detected previously saved calibration (autoInit) files. That's fine as that option is not released yet. 
The error is this: 
`[error] [STREAM]
Failed to resolve the request:
Format: Z16, width: 640, height: 480
Format: Y8, width: 640, height: 480
Into:
Formats:
Y8
Z16`
This means that the SDK failed to open those streams in the L515 device, probably because they're open in the viewer.
Try closing the viewer (both DWS and Librealsense).

rvermeire, 2020-08-31:
Hi yschleif,

Thanks for the quick update. Now indeed that error has gone, but it is still crashing. Maybe because following error that was also in the previous log:

> [2020-08-31 19:22:32.982] [DWSSDK] [error] [STREAM] Command not found!

Full console log:

> [2020-08-31 19:22:27.180] [DWSSDK] [info] [DWSSDK] LOG CREATED
[2020-08-31 19:22:27.181] [DWSSDK] [info] [DWSSDK] SDK created with mode=0
[2020-08-31 19:22:27.181] [DWSSDK] [info] [DWSSDK] OMP threads: 8
[2020-08-31 19:22:27.181] [DWSSDK] [info] [DWSSDK] SDK_create
[2020-08-31 19:22:27.182] [DWSSDK] [info] [DWSSDK] looking for files in: C:\*\static_mode_config.json
[2020-08-31 19:22:27.183] [DWSSDK] [info] [DWSSDK] Successfully parsed configuration
[2020-08-31 19:22:27.259] [DWSSDK] [info] [DWSSDK] Preparing background worker...
[2020-08-31 19:22:27.260] [DWSSDK] [info] [DWSSDK] Initializing player...
[2020-08-31 19:22:28.325] [DWSSDK] [info] [STREAM] Prepare LibRealsense terminal...
[2020-08-31 19:22:28.351] [DWSSDK] [info] [STREAM] Prepare LibRealsense terminal... DONE
[2020-08-31 19:22:29.961] [DWSSDK] [info] [STREAM] Streaming started with device: 00000000f0220767, FW: 01.04.01.02
[2020-08-31 19:22:32.688] [DWSSDK] [info] [DWSSDK] Initializing player... DONE
[2020-08-31 19:22:32.695] [DWSSDK] [info] [DWSSDK] LOG CREATED
[2020-08-31 19:22:32.695] [DWSSDK] [info] [STREAM] ======================
[2020-08-31 19:22:32.696] [DWSSDK] [info] [STREAM] Attempt to tun script: C:\*\logState.txt
[2020-08-31 19:22:32.697] [DWSSDK] [info] [STREAM] > irb e2 09 1  // laser modulation limit
[2020-08-31 19:22:32.729] [DWSSDK] [info] [STREAM] Offset: 0x0 => 2D
[2020-08-31 19:22:32.730] [DWSSDK] [info] [STREAM] > irb e2 0a 1  // laser modulation limit
[2020-08-31 19:22:32.754] [DWSSDK] [info] [STREAM] Offset: 0x0 => 2D
[2020-08-31 19:22:32.755] [DWSSDK] [info] [STREAM] > mrd a00e1890 a00e1894 // invalidation configuration
[2020-08-31 19:22:32.778] [DWSSDK] [info] [STREAM] Offset: 0x0 => 00000000
[2020-08-31 19:22:32.801] [DWSSDK] [info] [STREAM] > mrd a00e1600 a00e1604 // edges configuration
[2020-08-31 19:22:32.880] [DWSSDK] [info] [STREAM] Offset: 0x0 => 00000200
[2020-08-31 19:22:32.880] [DWSSDK] [info] [STREAM] > mrd a00e1894 a00e1898 // Confidence configuration
[2020-08-31 19:22:32.903] [DWSSDK] [info] [STREAM] Offset: 0x0 => 00000000
[2020-08-31 19:22:32.904] [DWSSDK] [info] [STREAM] > mrd a00e0b0c a00e0b10 // Sharpness configuration
[2020-08-31 19:22:32.928] [DWSSDK] [info] [STREAM] Offset: 0x0 => 00000028
[2020-08-31 19:22:32.928] [DWSSDK] [info] [STREAM] > mrd a0020ca8 a0020cac // Rast configuration
[2020-08-31 19:22:32.954] [DWSSDK] [info] [STREAM] Offset: 0x0 => 000000FF
[2020-08-31 19:22:32.955] [DWSSDK] [info] [STREAM] > apd_flyback_values_get // Get flyback values
[2020-08-31 19:22:32.979] [DWSSDK] [info] [STREAM] state:          5
isDataValid:    1
reserved16:     1900
vSenseReq:      F701
pwmValue:       DB03
tSenseRead:     3002
vSenseRead:     F901
tSenseC:        33.94037628
vSeneseReadV:   -148.0544128
reqV:           -148.2430267
[2020-08-31 19:22:32.980] [DWSSDK] [info] [STREAM] > RdCalibInfo 12
[2020-08-31 19:22:32.982] [DWSSDK] [error] [STREAM] Command not found!
[2020-08-31 19:22:32.983] [DWSSDK] [info] [STREAM] ======================
[2020-08-31 19:22:33.086] [DWSSDK] [info] [BM_API] No floor calibration data detected. Creating new initialization flow.

Kind regards,

Ronald

rvermeire, 2020-08-31:
Update:

When I run in debugging mode I still get an error: 

> Unhandled exception at 0x00007FFC04A9A719 in rs_dws_01.exe: Microsoft C++ exception: InferenceEngine::details::InferenceEngineException at memory location 0x000000721B0FD890

however the program can be continued.  It even calls DWS_Start, but then inmediatally call stops again. When I remove the UNKNOWN ERROR case temporarily I see this block in the log file:

> [2020-08-31 20:23:56.229] [DWSSDK] [info] [DWSSDK] Locking process Measure()...UNLOCKED
> new image received. ID = 732
> new image received. ID = 733
> [2020-08-31 20:23:56.300] [DWSSDK] [debug] [DWSSDK] measure() loop E
> [2020-08-31 20:23:56.311] [DWSSDK] [info] [DWSSDK] Waiting on camera input lock...
> [2020-08-31 20:23:56.312] [DWSSDK] [info] [DWSSDK] Waiting on camera input lock... FREE
> [2020-08-31 20:23:56.313] [DWSSDK] [info] [DWSSDK] Gyro data: 0.006981, 0.005236, -0.001745,
> [2020-08-31 20:23:56.315] [DWSSDK] [info] [DWSSDK] Locking process Measure()...
> [2020-08-31 20:23:56.316] [DWSSDK] [debug] [FLOW]  BoxMeasureFlow_process 1
> [2020-08-31 20:23:56.316] [DWSSDK] [debug] [FLOW]  BoxMeasureFlow_process 2
> [2020-08-31 20:23:56.317] [DWSSDK] [info] [FLOW] Camera view: SIDE
> [2020-08-31 20:23:56.325] [DWSSDK] [info] [OPENVINO] Wrapper used in the CALL_STATUS_FNC was not initialized.
> [2020-08-31 20:23:56.326] [DWSSDK] [warning] [DWSSDK] single frame error
> [2020-08-31 20:23:56.327] [DWSSDK] [debugSeverity: ] [DWSSDK] measure() loop S
> 1, Code: 12, single frame error
> [!] APP: 0 Boxes detected!

Kind regards,

Ronald

yschleif, 2020-08-31:
Hi Ronald,

This is not an issue:

> [2020-08-31 19:22:32.982] [DWSSDK] [error] [STREAM] Command not found!

We use OpenVINO as an inference engine, and from your log it seems it had a crash while initialising.
I'd guess the issue is OpenVINO not finding the dlls it's looking for where it expects them. 
If you placed your dlls in a place other than by your .exe, you'll have to put OpenVINO related dlls next to your binary.
namely:

> clDNNPlugin.dll
inference_engine_c_api.dll
inference_engine_legacy.dll
inference_engine_lp_transformations.dll
inference_engine_nn_builder.dll
inference_engine_transformations.dll
libmmd.dll
MKLDNNPlugin
ngraph.dll
svml_dispmd.dll
tbb.dll
tbbmalloc.dll

Try copying plugins.xml too for good measure

Joe

rvermeire, 2020-08-31:
Hi Joe, 

plugins.xml did the trick, that was all I needed to change. Thanks for your support!

Kind regards,

Ronald

closed 2020-08-31 by rvermeire