realsenseai/librealsense #7179: Pyrealsense2 convert depth pixel to color pixel?

mawin958700, 2020-08-24:
Model: Intel Realsense D435
Question: For example, I've depth pixel coordinate is [100,100] in depth_frame, and I need to convert it into Color pixel coordinate. I've found this function from documentation "rs2_project_color_pixel_to_depth_pixel(...)" <https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.html> how can I use this?

Thank for your reply.

MartyG-RealSense, 2020-08-25:
Hi @mawin958700   If your goal is to map a depth pixel coordinate to a color pixel coordinate, then the Python script in the link below sounds more like you want to do than using **rs2_project_color_pixel_to_depth_pixel** (mapping color to depth).

https://github.com/IntelRealSense/librealsense/issues/1890

The discussion warns that the results may be a bit "out of range" for some pixels when used with D435, due to differences between the color and depth FOV on that camera (whereas the D415 shouldn't be affected much as that model has the same FOV size for color and depth).

https://github.com/IntelRealSense/librealsense/issues/1890#issuecomment-397835880

mawin958700, 2020-08-25:
I think the problem was resolved, Thank for your suggestion.

MartyG-RealSense, 2020-08-25:
Great news - thanks very much for the update!

MartyG-RealSense, 2020-09-03:
Case closed due to no further comments received and problem being reported as resolved.

closed 2020-09-03 by MartyG-RealSense