realsenseai/librealsense #7180: Using L515 with Unreal Wrapper

roactheraven, 2020-08-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 01.04.01.02 | 
| Operating System & Version |   Win (10)   | 
| Platform                                 | PC |
| SDK Version                            |  2.36.0                         | 
| Language                            |  C++                        | 
| Segment			|  Unreal                          | 

### Issue Description
I've been trying to get the Unreal Wrapper to work but so far have been unsuccessful.  I'm sure there is some simple answer for what I'm missing, but so far when I hit Play in the demo level the cube's textures don't update from the checker pattern and I don't have a point cloud.  It seems to me like I'm not receiving the textures at all.  Is there some setting that I'm missing? I have the Enable Raw Depth, Enable Color, etc. all checked on.

[image]
[image]
[image]

Sashi-Mee, 2020-10-06:
hey ! 
Did you manage to make it work ?
I'm having issues using the 4.25, it won't open the project : it's asking me to rebuild the modules, but it fails and asks me to do it manually, but i have no idea how! 😢

roactheraven, 2020-10-06:
No I haven't managed to get it to work yet.  I figure it's something simple I'm just missing cause I wouldn't think I would have to dig too deep into the code.  I also haven't been able to devote a ton of time into it.

RealSenseSupport, 2020-10-08:
We are investigating Unreal support and L515 camera.

Sashi-Mee, 2020-10-09:
Okay thanks. It would be amazing that it'd work by next friday, as I participate in a hackaton weekend on moiton design and performing arts, and it would be a great tool to bridge reality and virtual worlds.

Thank you i andvance for the efforts !

roactheraven, 2020-11-18:
Have there been any updates for this?

manson, 2020-12-30:
The same problem I had. The problem occurred because of default camera resolution set is incompatible with L515 camera. Besides, this resolutions differ according to USB type the camera is attached through. Please, try this parameters in Inspector parameters to make it work: 
1) For USB3 - D:1024x768x30, I:1024x768x30, C:1920x1080x30 or D:320x240x30, I:320x240x30, C:960x540x6 
2) For USB2 - D:320x240x30, I:320x240x30 ,C:640x480x30
(D - depth config, I - infrared, C - color)

At this time there is an unaccepted pull request (https://github.com/IntelRealSense/librealsense/pull/8036) that besides the converting project to UE4.26 makes camera autoconfiguration. You can try this.

RealSenseSupport, 2021-01-13:
@roactheraven - Have you been able to test out the information offered by manson?

*If we don’t hear from you in 7 days, this issue will be closed.

roactheraven, 2021-01-13:
Yes, manually changing the camera resolution has fixed the issue for UE4.25

Thank you.

RealSenseSupport, 2021-01-14:
Glad that worked.

closed 2021-01-14 by RealSenseSupport

MatthewLKomar, 2021-08-13:
The default values don't work with the D455s on my end. 
To get it to work, I'm using (copy and paste the contents inside brackets into ue4): 
Depth Config: [ (Width=848,Height=480,Rate=30) ]
Color Config: [ (Width=1280,Height=720,Rate=30) ]
Infrared Config [ (Width=848,Height=480,Rate=30) ]