realsenseai/librealsense #7181: IR pattern is reflected in RGB image of RealSense D455

sho-ji, 2020-08-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version | Windows10  | 
| Platform                                 | PC  |
| SDK Version                            | v.2.36.0 | 

I have launched RealSense D455 RGB and stereo cameras using RealSense Viewe.
When I checked the RGB image, the IR pattern was reflected.
What should I do now?

MartyG-RealSense, 2020-08-25:
Hi @sho-ji  If the IR dot pattern is visible on the RGB image (which it should not be) then it can indicate that the **Laser Power** setting (found under the **Controls** section of the Viewer's Stereo Module options in the side-panel) is too high.  Reducing the Laser Power value can make the dots disappear from the RGB image, as their visibility is connected directly to the Laser Power value (higher value = more visible, lower value = less visible).

sho-ji, 2020-08-25:
Thank you for your answer.
It was improved by reducing the output to the laser oscillation threshold, but the IR dot pattern is still displayed in the RGB image.
Is it difficult to improve further?

MartyG-RealSense, 2020-08-25:
There is a Visual Preset that can remove the dot pattern but it currently only works for D415 models.  It is coming soon for D455 in a firmware update though.

https://github.com/IntelRealSense/librealsense/issues/7149

Until then, the IR Emitter can be disabled in the Viewer using instructions in the above link.

agrunnet, 2020-08-25:
Please clarify. Is the IR pattern visible in the third RHB imager that has the IR-cut filter? It should not be. 
Or are you saying it is visible in the left stereo RGB image that has no IR cut filter?

sho-ji, 2020-08-26:
Hi @MartyG-RealSense 
Thank you for your answer.
I am looking forward to updating the firmware.
Regarding disabling the IR emitter, I don't want to disable the IR emitter because that condition has a lot of depth loss.

Hi @agrunnet 
Thank you for your reply.
The IR pattern is visibled in the RGB camera color stream.
Share the image.
[image]
Lowering the laser power reduced the reflection of the IR pattern, but the IR pattern did not disappear.

MartyG-RealSense, 2020-09-03:
Hi @sho-ji   Do you require further assistance with this case, please?   Thanks!

sho-ji, 2020-09-03:
I would like to know if the IR pattern visible in the RGB image is normal.
I feel that the IR pattern visible in the RGB image is stronger than the RealSense D400 series up to now.
Is it possible to remove this IR pattern with a visual preset that will be implemented in the future?

MartyG-RealSense, 2020-09-03:
@sho-ji  I will refer your case to a member of the RealSense team who is looking at issues related to D455.

sho-ji, 2020-09-03:
Hi @MartyG-RealSense Thank you very much!

agrunnet, 2020-09-03:
@sho-ji thanks for image.  This helps clarify. The bleed of IR into the RGB should not be more than D435. We are looking into this in more detail.

sho-ji, 2020-09-04:
@agrunnet  Thank you for your reply.
I look forward to hearing your detailed results.

agrunnet, 2020-09-04:
@sho-ji our engineers will try to contact you directly.

sho-ji, 2020-09-04:
@agrunnet Thank you very much!

MartyG-RealSense, 2020-09-13:
Hi @sho-ji   If you have received direct assistance from RealSense engineers, can this case be closed now please or do you wish to keep it open longer?    Thanks!

sho-ji, 2020-09-13:
Hi @MartyG-RealSense  In that case, this case can be closed.

MartyG-RealSense, 2020-09-14:
Thanks very much for the confirmation!

closed 2020-09-14 by MartyG-RealSense