realsenseai/librealsense #7182: L515 - C#: Advanced Mode not Supported

siwidmer, 2020-08-25:
Hello, 
I'm integrating a L515 into our C# Framework. I'd like to import a JSON File I created in the 'Intel RealSense Viewer'.

`var pipe = new Pipeline();
PipelineProfile pipelineProfile = pipe.Start(_config);
var advancedDev = AdvancedDevice.FromDevice(pipelineProfile.Device);

if (advancedDev.AdvancedModeEnabled)
{
	advancedDev.JsonConfiguration = System.IO.File.ReadAllText("Settings.json");
}                   
`
This code exits always with an exception saying: "The Device does not support advanced mode". Is there a property to enable the advanced mode? In the viewer exporting and importing settings files are working neatly. Is there an issue with the C# implementation?

dorodnic, 2020-08-26:
Hi @siwidmer 
The L500 indeed does not support Advanced Mode interface (in same sense of 90-ish controls D400 exposes), its not limited to C# wrapper. You can configure L500 completely by setting sensor options explicitly using the values in JSON file.

bmegli, 2020-08-27:
Hi @siwidmer,

In C++ with L515 you may load the json with `rs2::serializable_device` (instead of AdvancedMode which it doesn't support).

See C++ class hierarchy [here](https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1serializable__device.html).

```C++
// C++ example (not C#)
rs2::pipeline_profile profile = pipe.get_active_profile();

if(!json.empty())
{
  cout << "loading settings from json:" << endl << json  << endl;
  auto serializable  = profile.get_device().as<rs2::serializable_device>();
  serializable.load_json(json);
}
```

Have a look if it would be possible from your C# framework.

siwidmer, 2020-08-27:
Thank you @bmegli for your hint!

I investigated the C#-wrapper. Unfortunately, I didn't find any methode integrating your C++ fonctionality.

siwidmer, 2020-08-27:
@dorodnic: Thank you for pointing this out. Then I'll set all values explicity.

RealSenseSupport, 2020-09-15:
Hi @siwidmer 

Is anything further needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-22 by RealSenseSupport