realsenseai/librealsense #7185: Realsense D435 Camera

johnkflynn, 2020-08-25:
| | |
|---|---|
|**librealsense**|2.36.0 RELEASE|
|**OS**|Windows|

Please provide a description of the problem
Hi,
Backend in rs2_create_device(info_list:000002319C04C6B0, index:0):
MFCreateDeviceSource(_device_attrs, &_source) returned: HResult 0x800706d9: "There are no more endpoints available from the endpoint mapper."

I get this error message when I run the viewer. Device manager recognises and installs the drivers for the camera D435, but it is not accessible either with the viewer or with RecFusion App. I'm running Version	10.0.19042 Build 19042
of windows. Thanks,
John

MartyG-RealSense, 2020-08-25:
Hi @johnkflynn   Could you first check please if access to the camera is set to On in the privacy settings of Windows 10, please?  Instructions for doing so are in the link below.

https://github.com/IntelRealSense/librealsense/issues/3381#issuecomment-503424789

johnkflynn, 2020-08-25:
> #3381 (comment)
> Hi, yes access to camera granted. I checked that a few times. I have
> uninstalled and reinstalled. When RealsenseViewer is starting it says
> camera found, but then hangs. On a rare occasion it gets into the programme
> it then says please connect a camera. All permissions for going through
> firewalls etc, granted. This started about 20 August.

John K Flynn

johnkflynn, 2020-08-25:
> The details in the case linked to below seem to be a close match to your
> situation:
>
> #3879 <https://github.com/IntelRealSense/librealsense/issues/3879>

Hi, thanks for all your help. I think I may have just solved it! When I
first installed it, it told me it needed an upgrade to the latest firmware.
At the time this was 05.12.06.00. I did this and problems started. I have
just now reverted to Firmware 05.11.06.200, the realsense viewer now works!
So does RecFusion! Again once again for the help.

John K Flynn

johnkflynn, 2020-08-26:
> Yes, problems can sometimes be corrected if you perform a firmware
> downgrade. You can then re-upgrade to the most recent firmware if you wish.
> Bear in mind that the camera sets a maximum limit of 20 firmware downgrades
> though before you can only upgrade to a newer firmware than the current one
> after that.
>
> Glad to hear that you found a solution!

O! Dear, Switched on the computer. Downloaded 3dot software. It found and
opened the camera. Then after a short while said it can't find the camera.
Intel Realsenseviewer  discovers the camera but then bombs out. Checked
obvious things the RS dfu service is running, drivers loaded just nothing
can relate to it. Sorry to bother you again.

John K Flynn

MartyG-RealSense, 2020-08-26:
It's no trouble at all.

It is a positive sign that the camera can at least be detected.  It sounds like a disconnection issue.  Are you using the official short USB cable supplied with the camera or a longer USB cable of your own choice, please?

johnkflynn, 2020-08-26:
Yes. It's the official one.I suspect it is a connection issue as well. I
have to go and do shopping. I'll be back later.

John K Flynn

johnkflynn, 2020-08-27:
A little bit more information I forgot to mention that I am running this on a Dell xps 13 7390 laptop. I suspect that is the problem. I connected the camera to the Desktop PC - no problem. It fired up right away. When I look at the setup on the laptop, the hub is connected by a thinner cable than the camera. While it can recognise the camera, I suspect it doesn't have enough power to drive the camera. I am awaiting a thicker c to c cable from Amazon. I will let you know how I get on.

MartyG-RealSense, 2020-08-27:
@johnkflynn  Thanks very much for the update - I look forward to your report on tests with the new cable.   Good luck!

johnkflynn, 2020-08-29:
Hi MartyG, I think I may have exhausted every option I have looked for. I bought the Dell docking station thinking a 100W would be great and would solve the problem! It helped to the extent RealsenseViewer recognised the D435 camera but then said the Metadata needs to be enabled on this commuter. I don't know how to do this. I tried Dot3D it found the camera but wouldn't or couldn't connect. I tried RecFusion who on starting showed a brief clip of what the camera was seeing, then nothing. I have removed and reinstalled a number of time. I have tried different cables. Would you know the names of the drivers I should have installed?. On my computer they are discovered as Intel Realsense Depth camera 435 with RGB Modual Depth. Driver =  Microsoft dated 21/06/2006, USB Video device. The second entry with the RGB Module RGB: there are 4 drivers ksthunk.sys: SecureUSBVideo.dll: usbvideo.sys: and WdmCompanionFiter.sys. On my main computer it looks like the drivers are from Intel. The Camera works with RecFusion on this computer!

MartyG-RealSense, 2020-08-29:
On Windows, you can enable camera metadata in the **RealSense Viewer** program.  If you launch the Viewer with the camera plugged in, you may get a pop-up message asking if you wish to enable metadata.  If you do not get the pop-up, unplugging the camera and plugging it back into a different USB port can trigger the message.

If you can get an initial image before disconnection occurs, it sounds as though the drivers are present and okay.  Here is an image of my own Device Manager with a D435i camera, with the drivers under the **Cameras** category.

[image]

johnkflynn, 2020-08-30:
Hi MartyG, When I run Realsense viewer I keep getting error messages. I
attach a screenshot. You have the same entries in the Camera section of
device manager. If you press the properties button what are the details of
the driver? Is it microsoft or realsens?

[image: image.png]

MartyG-RealSense, 2020-08-30:
Your drivers look fine to me.  The properties of my drivers are below:

[image]

[image]

johnkflynn, 2020-08-30:
Your drivers are from Microsoft mine are from Intel. See Screenshot

[image: image.png]

MartyG-RealSense, 2020-08-30:
I have the same date on my drivers as yours.  You have had the RealSense Viewer working before, so there likely is not a problem with the drivers that are installed.

[image]

johnkflynn, 2020-08-30:
OK. On my first screenshot there was an error shown in the viewer. Any idea
what is causing it?

John K Flynn

MartyG-RealSense, 2020-08-30:
None of the images in the replies that you are posting by email are appearing on the GitHub website unfortunately so I cannot see them.  When I said that I could "see" your drivers, I was referring to the text details of them posted earlier.  Apologies for the confusion.

https://github.com/IntelRealSense/librealsense/issues/7185#issuecomment-683272164

The **MFCreateDeviceSource** errors that you are experiencing are Windows errors and not RealSense errors.  So this suggests that the problem is with the Windows version on the affected computer and not from the RealSense SDK software.

The link below has some information that you can use as a starting point to investigate the problem.  There may not be anything that the RealSense team can do to correct a Windows error on your system that is not caused by RealSense.

https://thegeekpage.com/fix-there-are-no-more-endpoints-available-from-the-endpoint-mapper-error-in-windows-10/

johnkflynn, 2020-09-06:
Hi Marty,

I am still having problems. I have been looking at svchost.exe as the culprit, but no suggestions seems to work. One result is realsense viewer v2.38.1 stays open a little longer and asks the question do you want to turn on Advanced mode. When I say yes the viewer shut down. If I tried to switch on the stereo module I get this error message: 
[image]:

UNKNOWN in rs2_get_option(options:128C6C60, option:Backlight Compensation):
Device must be powered to query video_proc!

I have searched high and low to try and solve this. The D435 works perfectly on my Desktop, but on my Dell XPS 13 Lap top No Go. Any thoughts would be welcomed! Is there a command prompt I could run to switch on Advanced Mode, or Change the filters?

John

MartyG-RealSense, 2020-09-06:
As your problems apparently began at SDK version 2.36.0, it may be worth investigating whether the problem is related to a conflict with an **OpenVINO** function that was introduced into the Viewer in 2.36.0.  

This can be tested simply by downloading the 32.5 MB standalone version of the Viewer - **Intel.RealSense.Viewer.exe** - from the SDK's **Assets** list, as this version does not have the OpenVINO function.  Conflicts with the OpenVINO function that occur with the Viewer packaged with the SDK installation therefore do not occur in the standalone version.

You can download the standalone version for SDK 2.38.1 from here:

https://github.com/IntelRealSense/librealsense/releases/tag/v2.38.1

[image]

Once downloaded, it should be launched from the location that it was downloaded to and not launched using the desktop icon (which points to the original Viewer version that was installed with the SDK).  If you wish, you can create a desktop shortcut for the standalone Viewer executable.

johnkflynn, 2020-09-06:
This is the viewer I am using, and downloaded from where you said. When I select a preset or try to switch on the stereo mode this is the error message I get: 

Backend in rs2_get_option(options:110F1D10, option:Exposure):
hr returned: HResult 0xc00d36b6: "This object needs to be initialized before the requested operation can be carried out."

If I try to close the error box I need to press the close button about 15 - 18 times. I can then move around and change the viewing aspect of the desktop, but if I chose the RGB module it closes down the whole process.

MartyG-RealSense, 2020-09-06:
One way to eliminate whether the camera is responsible for the problems would be to download a sample bag file from the link below.  Once it is downloaded, unplug the camera, launch the Viewer and try drag and dropping the bag file into the center panel of the Viewer to load the file into the Viewer for playback.  

If you are able to change the options in the side panel without problems then this will suggest that something related to the camera being connected to the computer is causing problems.

https://github.com/IntelRealSense/librealsense/blob/master/doc/sample-data.md

johnkflynn, 2020-09-06:
I download the stairs bag file and started viewer. It load ok asking to connect a camera. I draged the Stairs. bag and got the following error.
[image]

johnkflynn, 2020-09-06:
I just tried it again and now it runs the bag file!

MartyG-RealSense, 2020-09-06:
Okay, great!  After the bag is loaded in, are you able to use the side-panel options available without an error occurring?

johnkflynn, 2020-09-06:
Yes I can access all modules post etc and no errors!

MartyG-RealSense, 2020-09-06:
Okay, thanks!  That would suggest that the problem is related to when the camera is connected.  Since the camera functions fine on your desktop computer, it is unlikely that the camera hardware, the camera firmware driver or the USB cable are at fault.  That would make it more likely that the problem is occurring on the USB port.

Do you have a different port on the laptop that you could try the camera in if you have not done so already (either USB 3 or USB 2 will be fine, since a D435 will work with both).

johnkflynn, 2020-09-06:
I have tried every  usb port 4 type C on the Laptop and also on the Dell Power connector. I'm beginning to wonder if this Dell xps 13 7390 is a bit too new? The resolution on the screen is 3840x2160. I have tried to lower all the resolutions but it made no difference.

MartyG-RealSense, 2020-09-06:
There was a recent case where the camera worked fine in a full size USB port but not in a USB-C port (if you mean you are plugging the camera into a USB-C micro size port on the laptop).  In that case, the RealSense user found that the full size USB ports had an Intel USB controller with xHCI support and the USB-C port that didn't work had a different USB controller without xHCI support.

https://github.com/IntelRealSense/librealsense/issues/6899

johnkflynn, 2020-09-06:
I think you are getting closer. This what I found in Device Manager. There two ports that hasxHCI support I have download the camera into each. No joy. When I went to the events viewer. I found they were not starting!  usbxhci.inf. Device PCI\VEN_8086&DEV_15D4&SUBSYS_09621028&REV_02\3936C0722DB3020000 had a problem starting.

Driver Name: usbxhci.inf
Class Guid: {36fc9e60-c465-11cf-8056-444553540000}
Service: USBXHCI
Lower Filters: 
Upper Filters: 
Problem: 0x15
Problem Status: 0x0

MartyG-RealSense, 2020-09-06:
Microsoft has a hidden hardware Troubleshooter option in Windows 10 that can scan for hardware problems and try to automatically correct them.

https://support.intelrealsense.com/hc/en-us/community/posts/360049169174/comments/360012666273

johnkflynn, 2020-09-07:
Well sorry to say it did not solve the problem. Uninstalling and reinstalling solved the problems of the drivers not starting. Both USBXHCI.inf and USBXHC.inf started. I ran the recommend realsense viewer 2.38.1 and got  this
[image]. I also ran Dota3d. It couldn't find the camera. Recfusion when it firsts loads it shows a snapshot of what the camera is looking at, but then refuses to show anything else. When you ask it to research it discovers the camera but the screen remains blank. The hidden troubleshooter didn't find a problem.

MartyG-RealSense, 2020-09-07:
I considered your latest information carefully.  If the D435 is not working properly with Recfusion or 3DOT, this makes me think that the problem may not be with the RealSense SDK software but with the camera drivers (the firmware driver or the RGB / depth drivers).   

Assuming that the firmware driver is okay, since it is stored inside the camera and it works on the desktop computer: that would leave the Depth and RGB drivers.  These are stored on the individual computer, so there is the potential there fore differences between the laptop and the desktop.

Could you open the **Device Manager** please, go to the **Camera** category.  Right-click on each of the two drivers and select **Uninstall device** to remove the driver.  When both of the drivers are uninstalled, unplug the camera, wait a couple of seconds and re-insert the camera.  This should reinstall the two drivers over your internet connection.

This will demonstrate whether your laptop had problems with its installed drivers.

[image]

johnkflynn, 2020-09-07:
I have done this a number of times now. I have just done once more. It first load Microsoft drivers 2006 version. Realsense viewer just hangs. After another few tries the Realsens Depth and RGB load up.

This allows Realsense viewer to recognise the camera, but won't go further because of an error. e.g. last message. If you press the ok button about 17 times it allows me to expand the sterio mode but not allow me to save. If I press the RGB it bombs out with this error 

|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Access Violation!

I have checked, as much as I can all the drivers loadedn Device Manager. I have a number of different dates on them. I have only had this laptop since July. I used Laplink to copy material from my old one. I wonder if I copied something that is screwing the drivers up? The installation date 27/07/2020. Some of the drivers are from about 20 August to 03/09/2020. I can't see this happening but when you are reaching the limits you try and grasp onto any straw that happens to flow past.

MartyG-RealSense, 2020-09-07:
As you have tried most other courses of action, you may as well try upgrading the camera firmware to the latest version (5.12.7.100), since this is the recommended firmware version for SDK 2.38.1.  

If that does not make a difference then I will escalate your case to my manager.

johnkflynn, 2020-09-07:
I’ve already tried this. Do you think that it is the latest processor or
other hardware on the dell XP’s 7390? Clutching at straws now.

John K Flynn

MartyG-RealSense, 2020-09-07:
The 400 Series cameras can work with any Intel processor.  There is the issues related to the Viewer and Depth Quality Tool on some computers in https://github.com/IntelRealSense/librealsense/issues/7080#issuecomment-671807292 but that would not account for the improper detection of the camera in software such as Recfusion .

Does your laptop otherwise work okay other than the camera problems and does not have problems with non-camera software?

johnkflynn, 2020-09-08:
This laptop is only about two months old. This is the first time I have
tried to connect a 3d camera. I have found an old r300 or r200 and I'm
going to try that to see what happens. I'll let you know.

MartyG-RealSense, 2020-09-08:
The newness of the computer should not make a difference.  Where problms occur where it works on one computer but not another, it can be related to the USB ports on the computer that it does not work with.  

Using an old R200 camera requires more setup than the modern cameras.  You have to install its drivers using a special Windows program called the Depth Camera Manager (DCM) and then install an obsolete SDK from 2016 to use with it, as the current SDK is not compatible with R200.

MartyG-RealSense, 2020-09-08:
A way to eliminate once and for all whether the laptop's USB ports are the problem may be to attach the camera to a mains electricity powered USB 3 hub.  The hub ports would have their own USB controllers separate from the laptop's, and be supplied with a stable electricity source from the mains socket, helping to avoid glitches related to instability of supplied power to the computer's built-in ports.

johnkflynn, 2020-09-08:
At the beginning we thought it was a power problem. I have a number of powered hubs and they didn't work. I bought the official Dell Dock WD19 180W and that didn't solve the problem. At the moment I am going through the whole computer getting rid of anything that hints of camera, realsense opengl etc. In device manager In the view tag I ticked show hidden items. Under cameras there were 8 mentions of D435 depth and RGB I removed them all. I am just about to insert the camera and see what happens.

MartyG-RealSense, 2020-09-08:
If you only have one camera attached then there should only be two D435 drivers listed under Camera - RGB and Depth.

johnkflynn, 2020-09-08:
You've probably heard a great shout of glee!! Its working! Both Dot3D and RecFusion. These are the steps I did. I uninstalled OpenNI 2.2 SDK prime sense. 
I uninstalled openGL. 
I uninstalled Realsense sdk v2.38.1.

In device manager I used the view tag to show hidden things. 
In the camera section 8 hidden entries appeared 4 for d435 Depth camera and 4 for d435 RGB I uninstalled them all.
 I didn't use realsense viewer to open the camera. I used dot3d.
 I will try realsense viewer2.38.1 later to see if it now works. I just wanted the pleasure of seeing the camera work! MartyG A great Big Thank You!!!

MartyG-RealSense, 2020-09-08:
Awesome news!   I wish you the best of luck with oyur further tests   :)

johnkflynn, 2020-09-08:
Just to bring everything up-to-date. I have tried the camera with 3 or 4 different scanning programmes, all ok. So I then turned to RealsenseViewer 2.38.1. The programme started without any error, but wouldn't allow me to turn on either the stereo module or the RGB module stopping with this error message: Invalid Value in rs2_open_multiple(sensor:103592C0, profiles:0EF75AD0, count:1):

Failed to resolve the request: 
	Format: Y8, width: 1280, height: 800

Into:
	Formats: 
	 Y8I

MartyG-RealSense, 2020-09-09:
Apologies for the delay in responding further.  I did tests with D415 and D435.  1280x800 was available in USB 3 mode but only as high as 1280x720 in USB2 mode.  So I suspect that you may have got the "Failed to resolve the request" message because your camera was in USB2 mode at the time.  

This error typically occurs when the SDK is asked to do something that it cannot carry out, perhaps due to a mistake in the stream definition that the SDK cannot understand or because it is a mode that this particular camera doesn't support at the time the request is made (e.g due to being in the more limited USB2 mode at the time)..

johnkflynn, 2020-09-09:
How would you switch it between between usb2 and 3? I only have usb 3 and 3.1. One of the things that I remembered was that this xps laptop 7390 comes with a fantastic scree.n When I checked just now it is running at 3840X2160. Perhaps this is what causes the problem. I was reading awhile ago, when you start an app the writing is so small on the screen that it is impossible to read. I read that this was because old Graphic code was compiled into the programme that didn't take advantage of the dynamic change in up-to-date resolutions. I did start to lower the resolutions but I am afraid I didn't continue down that path because I was side-tracked. I may give it another go.

MartyG-RealSense, 2020-09-09:
You do not need a USB2 port in order for the camera to be mis-detected as USB2.  Often, it can happen when the camera is in a USB3 port.  I believe it is related to how the USB controller is currently interpreting the state of the port.  

An example of mis-detection that can occur is when the camera's USB connector is inserted slowly instead of quickly and firmly, or if it jiggles slightly back out of the port (screw-lock USB cables can help to avoid this by securing the camera firmly in the port).

I haven't heard of librealsense being used on a screen with 3840X2160 resolution and there appears to be no information about anyone else having done so, unfortunately.  So I have no advice that I can offer about that.

MartyG-RealSense, 2020-09-18:
Hi @johnkflynn   Do you require further assistance with this case, please?   Thanks!

johnkflynn, 2020-09-19:
Hi MartyG,

Thanks for all your help.Everything seems to be working fine. I keep away
from the relsense viewer. Sometimes it works and other times it disconnects.
John

John K Flynn

MartyG-RealSense, 2020-09-19:
Okay, so is it okay to close this case please if you are satisfied that you do not need further help?   Thanks for the update!

MartyG-RealSense, 2020-09-29:
Case closed due to no further comments received.

closed 2020-09-29 by MartyG-RealSense

Freefly18, 2021-08-03:
This issue went through a lot, but I had the exact same error message as the original post and this was the only relevant Google result. As I've lost 2 weeks of my life finding the issue, I figured I would share the solution.

https://stackoverflow.com/questions/68638694/there-are-no-more-endpoints-available-from-the-endpoint-mapper-when-creating

TL;DR: Make sure the "Windows Camera Frame Server" service is running.

MartyG-RealSense, 2021-08-03:
Thanks so much @Freefly18 for sharing the solution that worked for you with the RealSense community!  I'm sorry to hear that you had such a difficult experience.