realsenseai/librealsense #7187: Images from D415 stereo camera a blurred. We cannot recognize bar codes when measuring from 70 cm distance.

Aaituov, 2020-08-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { R200 / F200 / SR300 / ZR300 / D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Win (8.1/10) / Linux (Ubuntu 14/16/17) / MacOS  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C/C#/labview/nodejs/opencv/pcl/python/unity }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

MartyG-RealSense, 2020-08-25:
Hi @Aaituov   Can you confirm which sensor on the camera is blurred please.   Is it the RGB color?  Thank you.

Aaituov, 2020-08-25:
@MartyG-RealSense Hi! Yep, RGB color. Output from camera which we get - JPEG image. Its quality is not satisfactory enough, so our/and thirt party proven object recognition algorithms could recognize bar code image.

MartyG-RealSense, 2020-08-25:
You can reduce blurring by using 60 FPS for the RGB.  Some more technical detail about this is on the link below:

https://github.com/IntelRealSense/librealsense/issues/3554

Alternatively you could look into whether the new RealSense D455 camera would be suitable for your project, as it has a fast global shutter for the first time like the depth sensor does (previous 400 Series models have used a slower rolling shutter on the RGB sensor).

Aaituov, 2020-08-26:
Hi! In #3554  they talk about blurred video image. While we have problems with photo image. Is there any solution for RGB blurred photoes?
Thank you!

MartyG-RealSense, 2020-08-26:
Blurring should still affect a static image if the observed object was moving quickly at the time that the single-frame image was captured.

If using 60 FPS for RGB is not an ideal anti-blurring solution for you, you could try reducing the RGB **exposure** value.  However, this darkens the image.  You can compensate by increasing the value of the RGB **gain**, though this may result in increased noise.

Aaituov, 2020-09-01:
Our object is static and moving. We didn't try to reduce RGB expose value and gain. BUt we increased FPS from 30 to 60. But quality of image became too bad for us. IT decreased from 1028 to 765.

MartyG-RealSense, 2020-09-01:
I ran tests where I tried to observe a normal sized bar code on a multipack of Coke using the RGB stream in the RealSense Viewer.  I can understand better now the results that you are getting.  At 70 cm away or more, the lines of the barcode start blurring together, making them likely unreadable.  Moving the camera in closer makes the lines of the code sharper.

Would it be possible to move the camera in closer than 70 cm in your project, please?  For example, if you are only using the RGB stream, the minimum depth restriction of the depth sensing does not apply.  When the depth stream is being used, you can move in as close as 0.3 meters on the D415, and closer if you reduce the camera's minimum distance by increasing the **Disparity Shift** value.

[image]

MartyG-RealSense, 2020-09-09:
Hi @Aaituov  Do you still require assistance wth this case, please?   Thanks!

Aaituov, 2020-09-09:
Hi! Yep we do!

Kind regards,
Askar Aituov
Connect on Twitter! <https://twitter.com/Aseke_Ai>

MartyG-RealSense, 2020-09-09:
Okay, thank you.  Let's recap on what the current situation is.  You want to scan bar codes and the camera is fixed at a distance of 70 cm from the bar code.  But the bar code is too blurred to recognize from that distance.  And putting some kind of magnifier in front of the RGB lens, as suggested above, is not an option.  Is that correct, please?

Aaituov, 2020-09-10:
Hi!
Thank you for your extra push to make this happen.
Yes, completely right.
We are training neural network (with help of Kerast library) to allow image
recognition for pictures with low resolution. We cannot increase power of
RGB camera.
Can you advice on flexible tuning of laser for this purposes?

Kind regards,
Askar Aituov
Connect on Twitter! <https://twitter.com/Aseke_Ai>

MartyG-RealSense, 2020-09-10:
I conducted further RGB tests with a D415 in the RealSense Viewer at 640x480 with a barcode and found that blurring of the barcode could be reduced by maximizing the **Sharpness** option to a value of '100'.  

Also, make sure that there is not a **Decimation** post-processing filter active on the RGB, as this can significantly increase blurring.

disabling RGB auto-exposure, maximizing RGB 'Gain' to 128 to brighten the image and reducing RGB 'White Balance' to around 3600 also helped to create a strong bright image that made the black and white of the bar code stand out.  The overall effect of all of these settings on the bar code is shown below.

[image]

Aaituov, 2020-09-11:
Thank you a lot.
It helped a bit, but didn't solve the problem.

Is there "auto-focus"  feature in Realsense camera?

Kind regards,
Askar Aituov
Connect on Twitter! <https://twitter.com/Aseke_Ai>

MartyG-RealSense, 2020-09-11:
There isn't an auto-focus on the RGB sensor of the RealSense cameras.  I ran further tests but the average size of the bar code is a limiting factor on what can be achieved at a 70 cm range due to the lines blending together at distance.

There is the possibility of using a separate non-RealSense RGB camera as the color image source, perhaps one with built-in auto-focus.  This may be impractical for your project though.

https://www.intelrealsense.com/depth-camera-external-sync-trigger/

closed 2020-09-29 by MartyG-RealSense