realsenseai/librealsense #7189: Realsense-viewer

rajanidulal, 2020-08-25:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {L515 } | 
| Operating System & Version |   {windows 10  | 
| Platform                                 | PC  |
| SDK Version                            |  2.0                         | 
| Language                            |  {C++ }                          | 

Issue Description
I am trying to run realsense-viewer (https://github.com/IntelRealSense/librealsense/tree/master/tools/realsense-viewer) in Visual Studio and it gave me "ux_windows is ambiguous" error. Even after defining the path, it gives "can not open include file: error" to some of the files. Do you have a sample project or a .sln file for this or a sub-project?

ev-mp, 2020-08-28:
@rajanidulal , can you be more specific on the steps you did and the (compilation?) error you receive with the tool?
As a general rule - in order to build and run SDK demos you need to start with configuring and running CMake at root directory.
That will generate the sln for Visual Studio.

Windows installation guide for building SDK from source - https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_windows.md

rajanidulal, 2020-09-01:
@ev-mp Thank you for your response. I tried making new project and it gave me a bunch of errors(too long to share it here, may have misconfigured it). When I tried to work on the project that you have on GitHub, I couldn't create a .sln file for this realsense-viewer. I ran CMake but it didn't help. Do you have steps or sets of configuration to create a .sln file to a project that is in tools folder? I can run the files you have in example folder but not files that are located in tools folder.

ev-mp, 2020-09-01:
@rajanidulal hello, 
Try to follow the build and installation guide as it is written to create the sln file for both the SDK and the Tools. 
There is no provisioning for sln file that will cover Tools directory only.
Just ensure that you select Librealsense's root directory in CMake's GUI for source code, `<root>/build` as target, and toggle on BUILD_EXAMPLES flag.

You can post CMake error log for review

rajanidulal, 2020-09-01:
@ev-mp Got it, thanks. I am trying to edit the viewer (like adding more functionality) in GUI. I see the source code and what I have here is working for the files that are examples file. Can you please direct me to the src code of the GUI only? I would like to deal with the working project of GUI. Thank you in advance. :D

ev-mp, 2020-09-06:
@rajanidulal , you can compile and run realsense-viewer using the steps described above.
The **sln** file you'd create will provide a `realsense-viewer` project entity that is compiled directly into EXE.
I'm not sure what do you mean by "GUI only" but all the source code files can be found under `<lrs_root>/tools/realsense-viewer/`.
The 3rd-party library used for the GUI is named ImGUI and is available under the `<lrs_root>/3rd-party/`

rajanidulal, 2020-09-10:
Sorry for the confusion, I was working on the viewer, that was what I meant by GUI. I may have missed somethings earlier but all my issues have been solved and it is working fine now. Thanks!

closed 2020-09-11 by rajanidulal