realsenseai/librealsense #7190: Error on Ubuntu 18.04

lmend22, 2020-08-25:
Error: Unrecoverable! error native function rs2_pipeline_start: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Inappropriate ioctl for device

I'm getting this error try to use the API on ubuntu 18.04 kernel 5.4

lmend22, 2020-08-25:
Camera is D435 
OS : ubuntu 18.04.5
SDK : 2.0
Language: Javascript

MartyG-RealSense, 2020-08-25:
Hi @lmend22   That error indicates a problem releating to the kernel.  There have been a couple of past cases where it appeared because a kernel was installed that was not yet supported by librealsense.

Whilst kernel 5.4 is not yet officially supported by librealsense (support only goes up to 5.3 officially at the time of writing this), numerous people have been able to use 5.4 successfully.

Which firmware version is your D435 camera using, please?

DGabriel12, 2020-08-25:
Hello, @MartyG-RealSense I am also getting the same error but I am using the D415 and my firmware version is 5.12.06

MartyG-RealSense, 2020-08-25:
Hi @DGabriel12   You could test whether your problem is related to the kernel by using a librealsense method called **RSUSB backend**.  It requires an internet connection but is not dependent on Linux versions or kernel versions and does not require patching.

https://github.com/IntelRealSense/librealsense/issues/6845#issuecomment-659540316

DGabriel12, 2020-08-25:
So I can open the real sense viewer this error I am getting comes when I try to run the examples from the https://www.npmjs.com/package/node-librealsense repo.

lmend22, 2020-08-25:
@MartyG-RealSense  I have tried kernels 4.16, 5.0, 5.3 and none of those worked

lmend22, 2020-08-25:
@MartyG-RealSense  how do I check the firmware driver?

MartyG-RealSense, 2020-08-26:
Hi @lmend22   Apologies for the delay in responding further.  You can check the firmware driver version with the command **rs-enumerate-devices**.   

It can also be checked in the RealSense Viewer program by clicking the Info button (the 'i' in a circle) at the top of the Viewer's options side-panel.

lmend22, 2020-08-26:
I checked the firmware and I have 5.12.6 so I don't think that the problem

lmend22, 2020-08-26:
@MartyG-RealSense any kernel that you know that definitely works?

MartyG-RealSense, 2020-08-26:
I would go for kernel 5.3 based on the fact that it is the most recent kernel that is officially validated for use with librealsense.  But I see that you have already tried that.

If you are having problems with multiple kernels then I would return to my earlier suggestion of bypassing the kernel by trying the RSUSB backend install.

https://github.com/IntelRealSense/librealsense/issues/7190#issuecomment-680300057

lmend22, 2020-08-26:
Can you provide any website that gives you instructions on how to do that?

MartyG-RealSense, 2020-08-26:
https://github.com/IntelRealSense/librealsense/issues/6845#issuecomment-659540316

lmend22, 2020-08-26:
@MartyG-RealSense  it didn't work :/

MartyG-RealSense, 2020-08-26:
What error did you get, please when using the RSUSB method?

lmend22, 2020-08-26:
the same one Error: Unrecoverable! error native function rs2_pipeline_start: set_xu(...). xioctl(UVCIOC_CTRL_QUERY) failed Last Error: Inappropriate ioctl for device

MartyG-RealSense, 2020-08-26:
You should not be getting kernel errors on an RSUSB backend installation, as the **-DFORCE_RSUSB_BACKEND=true** CMake instruction installs dependencies over the internet connection instead of having to patch them in (it is a no-patching-required process).  May I confirm please that you are not running the patch script after RSUSB installation has been completed.

lmend22, 2020-08-26:
I just installed the RSUSB I don't know what to do after that

MartyG-RealSense, 2020-08-26:
Once RSUSB installation is complete, you should not have to do anything else and should be able to use librealsense.

In the Ubuntu desktop, please try using the term **realsense viewer** in the search box to find the launch icon for the RealSense Viewer and try running it,

lmend22, 2020-08-26:
I think there is confusion here the real sense viewer works it has always worked I try to developing using the API for the real sense https://www.npmjs.com/package/node-librealsense using node js which is giving me that error.

MartyG-RealSense, 2020-08-26:
Okay, I understand.  I saw the mention of Javascript at the start of this case but it did not seem relevant, given that the problem was apparently focused on the kernel error (suggesting that it was purely a librealsense installation issue).  Thank you for the clarification.

Going over the case again from the start, I note that the librealsense SDK version that you are using was not mentioned.  Which version is it please?

lmend22, 2020-08-26:
@MartyG-RealSense 2.0

lmend22, 2020-08-26:
https://github.com/IntelRealSense/librealsense/issues/7190#issuecomment-680226679

MartyG-RealSense, 2020-08-26:
I mean the SDK version number.  For example, 2.36.0

lmend22, 2020-08-26:
it 2.36.0

MartyG-RealSense, 2020-08-26:
Thank you.  I asked because having a modern SDK helps to ensure that the problem is not related to an out of date SDK.  It is difficult to solve problems with node.js unfortunately due to a lack of past cases to refer to.

https://github.com/IntelRealSense/librealsense/issues/5856#issuecomment-591617064

My feeling is that if there is a problem related to the kernel then there may be something that went wrong or was missed in the node.js wrapper's installation process.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/nodejs#notice-before-you-go-further

MartyG-RealSense, 2020-09-03:
Hi @lmend22   Dyou still require asssistance with this case, please?  Thanks!

closed 2020-09-18 by MartyG-RealSense