realsenseai/librealsense #7191: is the custom calibraion only operated in window 10?

jasmin1026, 2020-08-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  python                        | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>
Hello, i read the guide for custom calibration, but as i see it can only be operated in windows 10? how can i get the intrinsic and extrinsic parameters for D435i under ubuntu 16.04 with python in OpenCV?

MartyG-RealSense, 2020-08-25:
Hi @jasmin1026   If you are referring to the **Dynamic Calibrator** tool, page 14 onward of the user guide fo the tool has instructions for installing it on Linux.

https://www.intel.com/content/www/us/en/support/articles/000026723/emerging-technologies/intel-realsense-technology.html

In regard to getting intrinsics and extrinsics with Python, the link below leads to a comment that has a script at the bottom of the comment that contains code for doing so.

https://github.com/IntelRealSense/librealsense/issues/1231#issuecomment-368421888

jasmin1026, 2020-08-25:
@MartyG-RealSense Thanks for your answer. 
I had read the programming guide for my own custom calibrator application, but in the guide only windows 10 is mentioned, i want to create the calibration for intrinsics (Especially for distortion）on Linux.  so i don't know whether it also works on Linux?And Dynamic Calibrator works only for extrinsics, it's not so useful for me.

MartyG-RealSense, 2020-08-25:
The Linux installation process for the tool installs both the Dynamic Calibrator and the Calibration API.

There are Linux details in the programming guide.  It is much easier to find them if you do a Find search in the document for the word **linux** so that you can browse quickly through the 20 references to that word in the programming guide.  That will lead you easily to Linux-specific information.

jasmin1026, 2020-08-26:
@MartyG-RealSense  ok, i've found it. thanks!

closed 2020-08-26 by jasmin1026

MartyG-RealSense, 2020-08-26:
Great to hear that you found what you needed - thanks for the update!