realsenseai/librealsense #7193: pyrealsense2: how to validate camera settings with every frame

Chandler, 2020-08-26:
Hello! I am starting to use advanced parameters of the realsense such as 
 * depth units
 * disparity shift
 * visual presets

In order to set these parameters, I need to be absolutely sure that the camera has been properly configured and has not reverted back to default values for some reason.

For example, **how can I validate the depth units of a frame every time I get a frame**. I need this because I have already run into issues where I thought I set the depth units but later found out that the data being returned did not have the units I expect.

I cannot figure out how to determine a frames depth units from here:
https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.depth_frame.html#pyrealsense2.depth_frame

https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.frame_metadata_value.html#pyrealsense2.frame_metadata_value

Thanks a lot.

MartyG-RealSense, 2020-08-26:
Hi @Chandler  In Pyrealsense2, code to get the current depth scale may look like this:

**depth_sensor = pipe_profile.get_device().first_depth_sensor()
depth_scale = depth_sensor.get_depth_scale()**

You could then run checks on the value of **depth_scale** and take an action.  This could include just printing out the value on-screen, or setting the depth_scale to 0.001 if it is not currently 0.001 by using a line such as:

**depth_sensor.set_option(rs.option.depth_units, 100.0)**

Chandler, 2020-08-26:
Thanks for the tip @MartyG-RealSense it works! I was hoping the information would literally be on the frame object but it will also suffice to query the device each time I get a frame.

Chandler, 2020-09-03:
@MartyG-RealSense actually I think this is still an issue.

Let's say I change the camera settings at time T1, and the I read a frame at time T2. How can I know if the new frame has the new settings? It would be nice to have a strong transnational guarantee that bytes read from the camera have the correct parameters.

As it is, there is some uncertainty in the seconds immediately after the settings change as to what the parameters of the data is. Which is a big deal for parameters like dept units.

MartyG-RealSense, 2020-09-03:
It can be assumed that when a setting is changed there is going to be some kind of transition time from the previous state to the new state, with the amount of time depending on the particular setting being changed.  I would recommend assuming that it could take several frames for a setting adjustment to complete.  

A RealSense team member provides more details about this in the link below.

https://github.com/IntelRealSense/librealsense/issues/3467#issuecomment-483410178

I am not personally aware of any way to guarantee that changes take place in the time of a single frame.

closed 2020-09-16 by Chandler