realsenseai/librealsense #7194: Incorrect velocity readings T265 Pose with Wheel Odometry.

meghdeepj, 2020-08-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.951 | 
| Operating System & Version |  Ubuntu 18.04 LTS, Linux4Tegra  | 
| Kernel Version (Linux Only)    |  5.4.0-42-generic | 
| Platform                                 | PC, Nvidia Jetson Nano |
| SDK Version                            |  v2.35.2                   | 
| Language                            | C/C++, ROS   | 
| Segment 			|     Robot   | 
| ROS Wrapper              |  v2.2.14   |

### Issue Description
I am trying to perform 2D odometry with T265 and wheel odometry for an autonomous mobile robot. The robot moves fairly slowly ~0.3m/s. The wheel odometry shows perfect 0 m/s when stationary and accurate in linear and angular velocity readings but accumulates significant drift in orientation (yaw) and X,Y pose and there is always a change of slippage. But, the twist data from wheel odometry fused T265 odom shows linear velocity of about 2mm/s when the robot is not moving (there are no vibrations when the robot is stationary) which is leading to growing X,Y drift even when the robot is stationary, but the twist data during motion is good. The orientation (yaw) from T265 pose is near perfect.

1. Is there any IMU calibration tool/ API/ package that can be used, so that it shows ~0 m/s when it should? 
2. Can vibrations during motion have an effect on T265's velocity readings when the motion has stopped? But this scenario is not relevant if the robot still hasn't started moving.
3. Any other way to solve this issue?

Below is the realsense-ros launch file contents:

      <launch>
  
      <rosparam>
          /rs_t265/tracking_module/enable_pose_jumping: false
      </rosparam>

       <arg name="serial_no"           default=""/>
       <arg name="json_file_path"      default=""/>
       <arg name="device_type"         default="t265"/>
       <arg name="camera"              default="rs_t265"/>
       <arg name="tf_prefix"           default="$(arg camera)"/>
       <arg name="initial_reset"       default="false"/>

       <arg name="fisheye_width"       default="848"/> 
       <arg name="fisheye_height"      default="800"/>
       <arg name="enable_fisheye1"     default="true"/>
       <arg name="enable_fisheye2"     default="true"/>
       <arg name="fisheye_fps"         default="30"/>
       <arg name="gyro_fps"            default="200"/>
       <arg name="accel_fps"           default="62"/>
       <arg name="enable_gyro"         default="true"/>
       <arg name="enable_accel"        default="true"/>
       <arg name="enable_pose"         default="true"/>
       <arg name="enable_sync"           default="false"/>

       <arg name="linear_accel_cov"      default="0.01"/>
       <arg name="unite_imu_method"      default="linear_interpolation"/>
       <arg name="publish_odom_tf"       default="true"/>
       <arg name="odom_frame_id"         default="odom"/>
       <arg name="topic_odom_in"         default="/odom"/>
       <arg name="calib_odom_file"       default="$(find loc_mod)/params/calibration_odometry.json" />"/>
         
       <arg name="base_frame_id"         default="base_link"/>  
       <arg name="pose_frame_id"         default="$(arg tf_prefix)_pose_frame"/>

       <node pkg="tf" type="static_transform_publisher" name="rs_t265_pose_to_base_link" args="0.0 -0.37 0.36 1.5708 1.5708 0.0 rs_t265_pose_frame base_link  30" />

       <group ns="$(arg camera)">
            <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
                <arg name="tf_prefix"                value="$(arg tf_prefix)"/>
                <arg name="serial_no"                value="$(arg serial_no)"/>
                <arg name="device_type"              value="$(arg device_type)"/>
                <arg name="json_file_path"           value="$(arg json_file_path)"/>
     
                <arg name="enable_sync"              value="$(arg enable_sync)"/>
                <arg name="fisheye_width"            value="$(arg fisheye_width)"/>
                <arg name="fisheye_height"           value="$(arg fisheye_height)"/>
                <arg name="enable_fisheye1"          value="$(arg enable_fisheye1)"/>
                <arg name="enable_fisheye2"          value="$(arg enable_fisheye2)"/>
                <arg name="fisheye_fps"              value="$(arg fisheye_fps)"/>
                <arg name="gyro_fps"                 value="$(arg gyro_fps)"/>
                <arg name="accel_fps"                value="$(arg accel_fps)"/>
                <arg name="enable_gyro"              value="$(arg enable_gyro)"/>
                <arg name="enable_accel"             value="$(arg enable_accel)"/>
                <arg name="enable_pose"              value="$(arg enable_pose)"/>

                <arg name="linear_accel_cov"         value="$(arg linear_accel_cov)"/>
                <arg name="initial_reset"            value="$(arg initial_reset)"/>
                <arg name="unite_imu_method"         value="$(arg unite_imu_method)"/>
                <arg name="publish_odom_tf"          value="$(arg publish_odom_tf)"/>
                 <arg name="topic_odom_in"            value="$(arg topic_odom_in)"/>
                 <arg name="calib_odom_file"          value="$(arg calib_odom_file)"/>
                <arg name="odom_frame_id"            value="$(arg odom_frame_id)"/>
                <arg name="base_frame_id"            value="$(arg base_frame_id)"/>
                <arg name="pose_frame_id"            value="$(arg pose_frame_id)"/>
         </include>
       </group>
 
     </launch>

And the "calibration_odometry.json" file:
`
{
    "velocimeters": [
        {
            "scale_and_alignment": [
                1.0,
                0.0,
                0.0,
                0.0,
                1.0,
                0.0,
                0.0,
                0.0,
                1.0
            ],
            "noise_variance": 0.0044,
            "extrinsics": {
                "T": [
                    0.0,      
                    -0.37,  
                    0.36    
                ],
                "T_variance": [
                    9.999999974752427e-7,
                    9.999999974752427e-7,
                    9.999999974752427e-7
                ],
                "W": [
                   -1.2091996,
		   1.2091996,
                    1.2091996
                ],
                "W_variance": [
                    9.999999974752427e-5,
                    9.999999974752427e-5,
                    9.999999974752427e-5
                ]
            }
        }
    ]
 }
`
Thanks for the help.

stevemartinov, 2020-09-02:
This is because the noise_variance is very small compareed to your T_variance of your wheel odometry. Of course the T265 will simply ignore all your odometry values

meghdeepj, 2020-09-08:
Thanks @stevemartinov, setting it to 0,02 has improved it, I'll tune it further.

closed 2020-09-08 by meghdeepj