realsenseai/librealsense #7198: D435 Device accessing fails

kremerf, 2020-08-26:
D435 Camera, Ubuntu 18.04, NVidia Jetson Nano, 05.12.05.00, librealsense version: 2.35.2, C++

After some time I got back to work with a previous project and unfortunately the access to the D435 is no longer working.

```
auto list = ctx.query_devices(); // Get a snapshot of currently connected devices
	if (list.size() == 0) 
	    throw std::runtime_error("No device detected. Is it plugged in?");
	rs2::device dev = list.front();
```

When I run this piece of code the program notices, that the list is of size !=0 (1 currently, which is the number of cameras connected). But as soon as it has to enter the list it crashes.

The code is from: https://dev.intelrealsense.com/docs/api-how-to I also looked at
https://github.com/IntelRealSense/librealsense/blob/master/examples/sensor-control/api_how_to.h#L4 and several other topics, that were similar.

```
[ERROR] [1598441678.950146804]: An error has occurred during frame callback: Error occured during execution of the processing block! See the log for more info
[ERROR] [1598441678.963453588]: main_processing.cpp || 1: Could not set Vars, will try again.
2RealSense error calling rs2_create_device(info_list:0x55c927f9c0, index:0):
    failed to set power state
```

```
try{
  ROS_DEBUG("main_processing.cpp || Setting Camera illumination time");
  ic.setExposure();
}
  catch (const rs2::error & e) {
  ROS_ERROR("main_processing.cpp || 1: Could not set Vars, will try again.");
  std::cerr << "2RealSense error calling " << e.get_failed_function() << "(" << e.get_failed_args() << "):\n    " << e.what() << std::endl;
  elay = seconds*_rate;
  return EXIT_FAILURE;
}
```

I also tried to generate a pipeline with looped devices, but no try has succeeded in reading the devicelist. I hope there is a simple error..

kremerf, 2020-08-26:
Power Supply is an actual "Power Supply EA-PS 2384-05 B" which is set to 5 volts and 4 amps, as the instructions from the jetson suggest. 

Opening the realsense-viewer reveals, that the camera is indeed working and connected via USB 3.2

MartyG-RealSense, 2020-08-26:
Hi @kremerf   Comparing your script for getting the first RealSense device to the version in the APi How To page, I noticed that you were missing line 1:

**rs2::context ctx;**

So that it should look like this:

`rs2::context ctx;
auto list = ctx.query_devices(); // Get a snapshot of currently connected devices
if (list.size() == 0) 
throw std::runtime_error("No device detected. Is it plugged in?");
rs2::device dev = list.front();`

kremerf, 2020-08-26:
Good perception.
I did not copy everything correctly - whoops.
But I just forgot it here, unfortunately, not in my script.

MartyG-RealSense, 2020-08-26:
Is the camera detected successfully if you use the command **rs-enumerate-devices**

kremerf, 2020-08-27:
Yes it is.

```
rs-enumerate-devices 
Device info: 
    Name                          : 	Intel RealSense D435
    Serial Number                 : 	936322072241
    Firmware Version              : 	05.12.05.00
    Recommended Firmware Version  : 	05.12.05.00
    Physical Port                 : 	2-1.2-3
    Debug Op Code                 : 	15
    Advanced Mode                 : 	YES
    Product Id                    : 	0B07
    Camera Locked                 : 	YES
    Usb Type Descriptor           : 	3.2
    Product Line                  : 	D400
    Asic Serial Number            : 	935723023874
    Firmware Update Id            : 	935723023874

Stream Profiles supported by Stereo Module
 Supported modes:
    stream       resolution      fps       format   
    Infrared 1	  1280x800	@ 30Hz	   Y8
    Infrared 2	  1280x800	@ 30Hz	   Y8
    Infrared 1	  1280x800	@ 25Hz	   Y16
    Infrared 2	  1280x800	@ 25Hz	   Y16
    Infrared 2	  1280x800	@ 15Hz	   Y16
    Infrared 1	  1280x800	@ 15Hz	   Y16
    Infrared 2	  1280x800	@ 15Hz	   Y8
    Infrared 1	  1280x800	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 30Hz	   Y8
    Infrared 1	  1280x720	@ 30Hz	   Y8
    Infrared 1	  1280x720	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 15Hz	   Y8
    Infrared 2	  1280x720	@ 6Hz	   Y8
    Infrared 1	  1280x720	@ 6Hz	   Y8
    Infrared 2	  848x480	@ 90Hz	   Y8
    Infrared 1	  848x480	@ 90Hz	   Y8
    Infrared 2	  848x480	@ 60Hz	   Y8
    Infrared 1	  848x480	@ 60Hz	   Y8
    Infrared 2	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 30Hz	   Y8
    Infrared 1	  848x480	@ 15Hz	   Y8
    Infrared 2	  848x480	@ 15Hz	   Y8
    Infrared 1	  848x480	@ 6Hz	   Y8
    Infrared 2	  848x480	@ 6Hz	   Y8
    Infrared 2	  848x100	@ 300Hz	   Y8
    Infrared 1	  848x100	@ 300Hz	   Y8
    Infrared 2	  848x100	@ 100Hz	   Y8
    Infrared 1	  848x100	@ 100Hz	   Y8
    Infrared 2	  640x480	@ 90Hz	   Y8
    Infrared 1	  640x480	@ 90Hz	   Y8
    Infrared 1	  640x480	@ 60Hz	   Y8
    Infrared 2	  640x480	@ 60Hz	   Y8
    Infrared 1	  640x480	@ 30Hz	   Y8
    Infrared 2	  640x480	@ 30Hz	   Y8
    Infrared 1	  640x480	@ 15Hz	   Y8
    Infrared 2	  640x480	@ 15Hz	   Y8
    Infrared 2	  640x480	@ 6Hz	   Y8
    Infrared 1	  640x480	@ 6Hz	   Y8
    Infrared 2	  640x400	@ 25Hz	   Y16
    Infrared 1	  640x400	@ 25Hz	   Y16
    Infrared 1	  640x400	@ 15Hz	   Y16
    Infrared 2	  640x400	@ 15Hz	   Y16
    Infrared 2	  640x360	@ 90Hz	   Y8
    Infrared 1	  640x360	@ 90Hz	   Y8
    Infrared 1	  640x360	@ 60Hz	   Y8
    Infrared 2	  640x360	@ 60Hz	   Y8
    Infrared 1	  640x360	@ 30Hz	   Y8
    Infrared 2	  640x360	@ 30Hz	   Y8
[... 261 lines cut ...]
There it is listed as the first Intel Corp. As that one is not there, when I disconnect the camera.
```
lsusb
Bus 002 Device 004: ID 8086:0b07 Intel Corp. 
Bus 002 Device 002: ID 0bda:0411 Realtek Semiconductor Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 8087:0a2b Intel Corp. 
Bus 001 Device 005: ID 0bda:8179 Realtek Semiconductor Corp. RTL8188EUS 802.11n Wireless Network Adapter
Bus 001 Device 004: ID 413c:8505 Dell Computer Corp. 
Bus 001 Device 002: ID 0bda:5411 Realtek Semiconductor Corp. 
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

```

And finally:

```
 v4l2-ctl --list-devices
Intel(R) RealSense(TM) Depth Ca (usb-70090000.xusb-1.2):
	/dev/video0
	/dev/video1
	/dev/video2

```
Seems to me, as the camera is best friends with the system as far as connection goes.

kremerf, 2020-08-27:
https://github.com/IntelRealSense/realsense-ros/issues/1036

I just read this one as I got the following error:

```
Running with LibRealSense v2.35.2
[ INFO] [1598513916.455881285]:  
[ INFO] [1598513916.727363467]: Device with serial number 936322072241 was found.

[ INFO] [1598513916.727520604]: Device with physical ID 2-1.2-4 was found.
[ INFO] [1598513916.727622116]: Device with name Intel RealSense D435 was found.
[ WARN] [1598513916.735640278]: Error extracting usb port from device with physical ID: 2-1.2-4
Please report on github issue at https://github.com/IntelRealSense/realsense-ros
[ INFO] [1598513916.735986063]: Resetting device...
[ INFO] [1598513922.777298399]:  
[ INFO] [1598513922.924441431]: Device with serial number 936322072241 was found.

[ INFO] [1598513922.924545287]: Device with physical ID 2-1.2-5 was found.
[ INFO] [1598513922.924595444]: Device with name Intel RealSense D435 was found.
[ WARN] [1598513922.929177786]: Error extracting usb port from device with physical ID: 2-1.2-5
Please report on github issue at https://github.com/IntelRealSense/realsense-ros

```

and in there was some PR#1088 fix, but that was from 2018, so I guess it is integrated in my software. Might this be the actual error we need to fix?

MartyG-RealSense, 2020-08-27:
May I confirm whether or not you are using the RealSense ROS wrapper to use ROS with librealsense, please?   The test scripts that you have been using are librealsense SDK scripts that are unrelated to ROS.  But your most recent message above is a ROS log, indicating that you are using ROS.  Please pardon my confusion.   :)

kremerf, 2020-08-27:
I use ROS yes, and for some of my debug-messages I also use the ROS-Logger to make it look uniform.

kremerf, 2020-08-27:
https://github.com/IntelRealSense/librealsense/issues/6347
https://github.com/IntelRealSense/librealsense/issues/6307

These two are also my issues and the only thing changed is, that I switched from the NUC to a Jetson Nano Developer Kit. So maybe there are some further questions already answered there, but in doubt feel free to ask them again.

I am currently installing the Jetpack 4.4 on another SD-Card to make debugging and error solving more convenient- If there is some progress there with this problem - when I get to it - I will also inform this topic.

MartyG-RealSense, 2020-08-27:
When using ROS, I would recommend the test of adding the term **initial_reset:=true** to the roslaunch statement if you are not using it already so that the camera performs a hardware reset on start-up.  This may sometimes yield a small improvement in performance, even if it does not always solve problems completely,

kremerf, 2020-08-28:
The initial reset is set to true.

```
 * /D435/realsense2_camera/accel_fps: 250
 * /D435/realsense2_camera/accel_frame_id: D435_accel_frame
 * /D435/realsense2_camera/accel_optical_frame_id: D435_accel_optica...
 * /D435/realsense2_camera/align_depth: True
 * /D435/realsense2_camera/aligned_depth_to_color_frame_id: D435_aligned_dept...
 * /D435/realsense2_camera/aligned_depth_to_fisheye1_frame_id: D435_aligned_dept...
 * /D435/realsense2_camera/aligned_depth_to_fisheye2_frame_id: D435_aligned_dept...
 * /D435/realsense2_camera/aligned_depth_to_fisheye_frame_id: D435_aligned_dept...
 * /D435/realsense2_camera/aligned_depth_to_infra1_frame_id: D435_aligned_dept...
 * /D435/realsense2_camera/aligned_depth_to_infra2_frame_id: D435_aligned_dept...
 * /D435/realsense2_camera/allow_no_texture_points: False
 * /D435/realsense2_camera/base_frame_id: D435_link
 * /D435/realsense2_camera/calib_odom_file:
 * /D435/realsense2_camera/clip_distance: -2.0
 * /D435/realsense2_camera/color_fps: 6
 * /D435/realsense2_camera/color_frame_id: D435_color_frame
 * /D435/realsense2_camera/color_height: 720
 * /D435/realsense2_camera/color_optical_frame_id: D435_color_optica...
 * /D435/realsense2_camera/color_width: 1280
 * /D435/realsense2_camera/depth_fps: 6
 * /D435/realsense2_camera/depth_frame_id: D435_depth_frame
 * /D435/realsense2_camera/depth_height: 720
 * /D435/realsense2_camera/depth_optical_frame_id: D435_depth_optica...
 * /D435/realsense2_camera/depth_width: 1280
 * /D435/realsense2_camera/device_type:
 * /D435/realsense2_camera/enable_accel: False
 * /D435/realsense2_camera/enable_color: True
 * /D435/realsense2_camera/enable_depth: True
 * /D435/realsense2_camera/enable_fisheye1: True
 * /D435/realsense2_camera/enable_fisheye2: True
 * /D435/realsense2_camera/enable_fisheye: False
 * /D435/realsense2_camera/enable_gyro: False
 * /D435/realsense2_camera/enable_infra1: True
 * /D435/realsense2_camera/enable_infra2: False
 * /D435/realsense2_camera/enable_pointcloud: True
 * /D435/realsense2_camera/enable_sync: False
 * /D435/realsense2_camera/filters:
 * /D435/realsense2_camera/fisheye1_frame_id: D435_fisheye1_frame
 * /D435/realsense2_camera/fisheye1_optical_frame_id: D435_fisheye1_opt...
 * /D435/realsense2_camera/fisheye2_frame_id: D435_fisheye2_frame
 * /D435/realsense2_camera/fisheye2_optical_frame_id: D435_fisheye2_opt...
 * /D435/realsense2_camera/fisheye_fps: 30
 * /D435/realsense2_camera/fisheye_frame_id: D435_fisheye_frame
 * /D435/realsense2_camera/fisheye_height: 720
 * /D435/realsense2_camera/fisheye_optical_frame_id: D435_fisheye_opti...
 * /D435/realsense2_camera/fisheye_width: 1280
 * /D435/realsense2_camera/gyro_fps: 400
 * /D435/realsense2_camera/gyro_frame_id: D435_gyro_frame
 * /D435/realsense2_camera/gyro_optical_frame_id: D435_gyro_optical...
 * /D435/realsense2_camera/imu_optical_frame_id: D435_imu_optical_...
 * /D435/realsense2_camera/infra1_frame_id: D435_infra1_frame
 * /D435/realsense2_camera/infra1_optical_frame_id: D435_infra1_optic...
 * /D435/realsense2_camera/infra2_frame_id: D435_infra2_frame
 * /D435/realsense2_camera/infra2_optical_frame_id: D435_infra2_optic...
 * /D435/realsense2_camera/infra_fps: 6
 * /D435/realsense2_camera/infra_height: 720
 * /D435/realsense2_camera/infra_width: 1280
 * **/D435/realsense2_camera/initial_reset: True**
 * /D435/realsense2_camera/json_file_path:
 * /D435/realsense2_camera/linear_accel_cov: 0.01
 * /D435/realsense2_camera/odom_frame_id: D435_odom_frame
 * /D435/realsense2_camera/pointcloud_texture_index: 0
 * /D435/realsense2_camera/pointcloud_texture_stream: RS2_STREAM_COLOR
 * /D435/realsense2_camera/pose_frame_id: D435_pose_frame
 * /D435/realsense2_camera/pose_optical_frame_id: D435_pose_optical...
 * /D435/realsense2_camera/publish_odom_tf: True
 * /D435/realsense2_camera/publish_tf: True
 * /D435/realsense2_camera/rosbag_filename:
 * /D435/realsense2_camera/serial_no:
 * /D435/realsense2_camera/tf_publish_rate: 0.0
 * /D435/realsense2_camera/topic_odom_in: odom_in
 * /D435/realsense2_camera/unite_imu_method:
 * /D435/realsense2_camera/usb_port_id:
 * /applicator/ApplNow1: 0.0
 * /applicator/angleDigi0: 0.0
 * /applicator/angleDigi1: 0.0
 * /applicator/newDest0: 0.0
 * /applicator/newDest1: 0.0
 * /applicator/vApplNow0: 0.0
 * /nuc/setCamera: True
 * /robot_name: paytah
 * /rosdistro: melodic
 * /rosversion: 1.14.9

```

kremerf, 2020-08-28:
I tested the following code-snippet with its two variations (**p.start()** and **p.start(cfg)**)

```
rs2::context ctx;
	auto list = ctx.query_devices(); // Get a snapshot of currently connected devices
    std::cout << "Found " << list.size() << " device(s)" << std::endl;
    
    
    rs2::pipeline p(ctx);
    rs2::config cfg;
    cfg.enable_device(serial_number);

    p.start();
    **or**
    p.start(cfg);

    rs2::device dev = p.get_active_profile().get_device();
    for (int i = 0; i < 2; i++) {
        auto sensor = dev.query_sensors()[i];
        sensor.set_option(RS2_OPTION_EXPOSURE, 2);
        std::cout << i << ": " << sensor.get_info(RS2_CAMERA_INFO_NAME) << std::endl;
        std::cout << i << ": " << sensor.get_option(RS2_OPTION_EXPOSURE) << std::endl;
        std::cout << i << ": " << dev.get_info(RS2_CAMERA_INFO_SERIAL_NUMBER) << std::endl;
    }
    p.stop();
```

And the output varies:

In the case of **p.start()**:

```
Start
Found 2 device(s)
0: Stereo Module
0: 2
0: 740112071197
1: RGB Camera
1: 2
1: 740112071197

```

In the case of **p.start(cfg);**:
```
Start
Found 2 device(s)
2020-08-28 08:04:09,794 WARNING [default] Could not open device failed to set power state
2020-08-28 08:04:09,994 WARNING [default] Could not open device failed to set power state
2020-08-28 08:04:15,005 WARNING [default] Could not open device failed to set power state
2020-08-28 08:04:15,262 WARNING [default] Could not open device failed to set power state
2020-08-28 08:04:15,398 WARNING [default] Could not open device failed to set power state
2020-08-28 08:04:20,398 WARNING [default] Could not open device failed to set power state
2020-08-28 08:04:20,639 WARNING [default] Could not open device failed to set power state
2020-08-28 08:04:20,771 WARNING [default] Could not open device failed to set power state
2020-08-28 08:04:25,780 WARNING [default] Could not open device failed to set power state
[ERROR] [1598594665.949696179]: main_processing.cpp || 1: Could not set Vars, will try again.
2RealSense error calling rs2_pipeline_start_with_config(pipe:0x557b33f000, config:0x557b5fa0e0):
    No device connected

```

The **cfg** element is just the Serial_Number, which is a simple **String**.
I found **void | enable_device (const std::string &serial)** on https://intelrealsense.github.io/librealsense/doxygen/classrs2_1_1config.html so I guess that is correct..

Seems odd. So I tested it with both Serial_Numbers and I could just find the one. Then I switched the USB-Ports and suddenly I found the other one - but just that one.
It seems as if currently just one single USB-Port (even though they are all identical) is viable to find a connected board via **cfg**.

Then in my last try, I killed the code with CTRL+C as per usual, and after all the ** .. killing on exit** it showed the second Serial_Number

```
Start
Found 2 device(s)
2020-08-28 08:22:19,511 WARNING [default] Could not open device failed to set power state
0: Stereo Module
0: 2
0: 936322072241
1: RGB Camera
1: 2
1: 936322072241
2020-08-28 08:22:20,126 WARNING [default] Could not open device failed to set power state
2020-08-28 08:22:20,301 WARNING [default] Could not open device failed to set power state
[detector_depth-6] killing on exit
[detector_rgb-5] killing on exit
[detector_processing-4] killing on exit
[D435/realsense2_camera-3] killing on exit
[D435/realsense2_camera_manager-2] killing on exit
terminate called without an active exception
0: Stereo Module
0: 2
0: 740112071197
1: RGB Camera
1: 2
1: 740112071197
[rosout-1] killing on exit
[master] killing on exit
shutting down processing monitor...
... shutting down processing monitor complete
done

```

So... it is able to retrieve the data?? I am very confused. But some progress in form of information has been made..

MartyG-RealSense, 2020-08-28:
Your code for getting the serial numbers from a multiple camera setup looks okay to me   

If you are using both librealsense and ROS, try not to have both of them trying to access the camera at the same time, as that could lead to a camera being "busy" as one program claims it and the other tries to access it.

When using ROS, an alternative to using serial numbers to distinguish between two different cameras is to identify them by USB port instead with **USB_PORT_ID**.  This is a function that was created during a period when a number of people were having trouble with multiple-camera launching.

https://github.com/IntelRealSense/realsense-ros#launch-parameters

Also, you can use a roslaunch instruction called **rs_multiple_devices.launch** that allows you to specify the serial numbers in the launch statement.

https://github.com/IntelRealSense/realsense-ros#work-with-multiple-cameras

kremerf, 2020-08-31:
The file rs_multiple_devices.launch is as follow:

```
<launch>
  <arg name="serial_no_camera1"    			default="936322072241"/> 			<!-- Note: Replace with actual serial number -->
  <arg name="serial_no_camera2"    			default="740112071197"/> 			<!-- Note: Replace with actual serial number -->
  <arg name="camera1"              			default="camera1"/>		<!-- Note: Replace with camera name -->
  <arg name="camera2"              			default="camera2"/>		<!-- Note: Replace with camera name -->
  <arg name="tf_prefix_camera1"         default="$(arg camera1)"/>
  <arg name="tf_prefix_camera2"         default="$(arg camera2)"/>
  <arg name="initial_reset"             default="true"/>

  <group ns="$(arg camera1)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="serial_no"             value="$(arg serial_no_camera1)"/>
      <arg name="tf_prefix"         		value="$(arg tf_prefix_camera1)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
      <arg name="usb_port_id"         value="2-1.1-3"/>
    </include>
  </group>

  <group ns="$(arg camera2)">
    <include file="$(find realsense2_camera)/launch/includes/nodelet.launch.xml">
      <arg name="serial_no"             value="$(arg serial_no_camera2)"/>
      <arg name="tf_prefix"		          value="$(arg tf_prefix_camera2)"/>
      <arg name="initial_reset"         value="$(arg initial_reset)"/>
      <arg name="usb_port_id"         value="2-1.1-5"/>
    </include>
  </group>
</launch>

```

On launch I get an error. I am a litle confused as what the usb_port_id actually is.
lsusb results in:

```
~/catstereo$ lsusb
Bus 002 Device 005: ID 8086:0b07 Intel Corp.
Bus 002 Device 003: ID 8086:0b07 Intel Corp.
```
So I got my values in the upper code-block from here (2-1.1-3 and 2-1.1-5) Is this correct or are they named differently?

```
RealSense ROS v2.2.12
[ INFO] [1598859163.268760156]: Running with LibRealSense v2.35.2
[ INFO] [1598859163.270293843]: RealSense ROS v2.2.12
[ INFO] [1598859163.270399523]: Running with LibRealSense v2.35.2
[ INFO] [1598859163.345874860]:
[ INFO] [1598859163.348468941]:
[ INFO] [1598859163.367935721]: Started >Depth<-ROS-Node
 31/08 09:32:43,540 ERROR [547096600960] (handle-libusb.h:95) failed to claim usb interface: 0, error: RS2_USB_STATUS_BUSY
terminate called after throwing an instance of 'rs2::error'
  what():  failed to set power state
[ INFO] [1598859163.623761542]: Device with serial number 936322072241 was found.

[ INFO] [1598859163.623876806]: Device with physical ID 2-1.1-3 was found.
[ INFO] [1598859163.623963267]: Device with name Intel RealSense D435 was found.
[ERROR] [1598859163.631678477]: Error extracting usb port from device with physical ID: 2-1.1-3
Please report on github issue at https://github.com/IntelRealSense/realsense-ros
[ERROR] [1598859163.631776396]: Please use serial number instead of usb port.
[ INFO] [1598859163.811865519]: Device with serial number 740112071197 was found.

[ INFO] [1598859163.811973231]: Device with physical ID 2-1.2-5 was found.
[ INFO] [1598859163.812058598]: Device with name Intel RealSense D435 was found.
[ERROR] [1598859163.816506479]: Error extracting usb port from device with physical ID: 2-1.2-5
Please report on github issue at https://github.com/IntelRealSense/realsense-ros
[ERROR] [1598859163.816608982]: Please use serial number instead of usb port.
[ERROR] [1598859163.829339172]: The requested device with serial number 740112071197 and usb port id 2-1.1-5 is NOT found. Will Try again.
```

MartyG-RealSense, 2020-08-31:
The link below has a long discussion **about usb_port_id** and overcoming the problem of getting **Error extracting usb port from device with physical ID**.

https://github.com/IntelRealSense/realsense-ros/pull/1088#issuecomment-596082480

kremerf, 2020-09-01:
Both of the packages they purged are already gone on my system.

```
sudo apt purge ros-melodic-realsense2*
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'ros-melodic-realsense2-description' for glob 'ros-melodic-realsense2*'
Note, selecting 'ros-melodic-realsense2-camera' for glob 'ros-melodic-realsense2*'
Note, selecting 'ros-melodic-realsense2-camera-dbgsym' for glob 'ros-melodic-realsense2*'
Package 'ros-melodic-realsense2-camera' is not installed, so not removed
Package 'ros-melodic-realsense2-camera-dbgsym' is not installed, so not removed
Package 'ros-melodic-realsense2-description' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.

sudo apt purge ros-melodic-librealsense2
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'ros-melodic-librealsense2' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.

```

MartyG-RealSense, 2020-09-01:
I note that in the multiple camera launch file, the camera names are set by default as Camera1 and Camera2, but they have a note beside them saying to "replace with camera name".  The name given for the cameras by the rs-enumerate-devices log that you kindly provided near the beginning of this case is **Intel RealSense D435**

After plugging the cameras in, it may be worth leaving the cameras in the same port and checking their physical ID in rs-enumerate-devices, then entering the ID number into the launch file, making sure not to change the cameras into differentUSB ports.

MartyG-RealSense, 2020-09-09:
Hi @kremerf   Do you still require assistance with this case, please?   Thanks!

kremerf, 2020-09-10:
I got advised from some colleagues to use the Jetson Xavier NX instead. As the nano seems to have too many little problems like this one.

So I unfortunately don't have a solution, but as for now it also isn't a problem for me anymore.

PS: I tried renaming them from Camera1 and Camera2 to Intel RealSense D435, but the names have to be unique and if I remember correctly, the names have been identical within `rs-enumerate`.

closed 2020-09-10 by kremerf