realsenseai/librealsense #7199: t265 imu problem

lishanggui, 2020-08-26:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { t265 } | 
| Firmware Version                   |0.2.0.951 | 
| Operating System & Version |   Ubunut 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-112-generic                                    | 
| Platform                                 | PC |
| SDK Version                            | librealsense SDK 2.36    | 
| Language                            | C++                       | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description

The sensor of T265 (including IMU) is calibrated on the production line, so no further calibration process is required (unlike the IMU on D435i). However, realsense-viewer shows that the acceleration will vary from 8 to 10 when placed at rest in different directions. Does this have an impact on vio and how to solve it?

lishanggui, 2020-09-25:
Hi @dorodnic can any people help me? thanks!

hannibal051, 2022-02-23:
As far as I know, D435i has the same problem. D435i and T265 share the same imu. After I do rs-imu-calibration on D435i, its acceleration varies from 9.7~9.9, before calibration the range is 8~10. So I guess if I can modify the rs-imu-calibration tool to run on T265, then the problem will be handled. Will update this comment.

davesmivers, 2024-10-30:
@youwyu did it work?

closed 2026-10-23 by MartyG-RealSense