realsenseai/librealsense #7200: qtcreator error on qmake

amirreza7876, 2020-08-26:
- ubuntu 16.04

- librealsense R200

i have installed qtcreator, when i click on run qmake this line shown up...
`SOFT ASSERT: "qmakeProject" in file /build/qtcreator-jrmbIB/qtcreator-3.5.1+dfsg/src/plugins/qmakeprojectmanager/qmakeprojectmanager.cpp, line 182
`
what did i do wrong?

MartyG-RealSense, 2020-08-26:
Hi @amirreza7876   Are you using a RealSense R200 camera with **QT Creator** and with the **librealsense** SDK please?

If so, please bear in mind that an earlier version of librealsense that is now known as **Legacy Librealsense** should be used with the R200 if you are not using it already.  This is because the current **librealsense2** SDK - also known as RealSense SDK 2.0 - is not compatible with the R200.  Legacy Librealsense can be found at the link below:

https://github.com/IntelRealSense/librealsense/tree/v1.12.1

Intel published a tutorial guide in 2016 for using the R200 and Legacy Librealsense with QT Creator.  This may be a useful reference to compare against the setup method that you used.

https://software.intel.com/content/www/us/en/develop/articles/using-librealsense-and-pcl-to-create-point-cloud-data.html

amirreza7876, 2020-08-26:
Hello @MartyG-RealSense thanks for your answer.
i have followed legacy repo and it says that i should install qtcreator...
in this line -- From the QtCreator top menu: Clean => Run QMake => Build -- i do the same but when i click run qmake the error came up.

MartyG-RealSense, 2020-08-26:
Okay, let's make sure that you are using the version of the QT Creator instructions from Legacy Librealsense and not the current RealSense SDK 2.0.  I've posted the Legacy SDK instructions for QT Creator below.

*************

sudo apt-get install qtcreator

sudo scripts/install_qt.sh (we also need qmake from the full qt5 distribution)

all.pro contains librealsense and all example applications

From the QtCreator top menu: Clean => Run QMake => Build

Built QtCreator projects will be placed into ./bin/debug or ./bin/release

amirreza7876, 2020-08-27:
thanks for your help @MartyG-RealSense

closed 2020-08-27 by amirreza7876

MartyG-RealSense, 2020-08-27:
You are very welcome @amirreza7876 - thanks for the update!