realsenseai/librealsense #7201: Object Pool Size

amitelle1337, 2020-08-26:
| Required Info                         |                                                |
| --------------------------------|-------------------------------- |
| Camera Model                       | D435i & L515 | 
| Firmware Version                   | 05.12.06.00 & 01.04.01.02 respectivly | 
| Operating System & Version | Win 10 | 
| Platform                                 | PC |
| SDK Version                           | 2.36.0 | 
| Language                               | C# | 
| Segment                                | Scanning | 

### Issue Description
I have a system with 6 cameras that capture synchronously (The system is for scanning an object from 6 angles).
I am trying to keep capturing time to a minimum (To prevent problems from noise and movement).
For that, I want to first capture from all the cameras and then start the processing part.
The issue I am having is that the object pool that Librealsense uses is too small for my needs: from my testing, I can only have 16 640X480 DepthFrames at once.
I would like to know if there's a way for me to increase the pool's size or at least know more information about it so I can adjust my program accordingly.
One solution I have in mind regarding whether I can change the pool's size or not is to split the capturing to more devices (For example, 6 Raspberry Pis - one for each camera), but that would be outside the scope of my project, unfortunately.
Another solution is to capture from each camera in a different process, thus each camera has its own pool, but that's assuming the object pool is not global, which I am lacking the information about.
Thanks in Advance for any help.

dorodnic, 2020-08-27:
Hi @amitelle1337 
Frame object has Keep method to indicate the frame is intended to be saved, and should no longer be part of the pool. 
Please note you need to call Keep on individual frames as well as the FrameSet object to make sure all pools are free. (Similar to how you call Dispose)

amitelle1337, 2020-08-27:
Thank you @dorodnic .
Is it possible to call `Keep` on individual frames but `Dispose` on the FrameSet object? Because after I retrieve the individual frames I don't really need the FrameSet object.

dorodnic, 2020-08-31:
Hi @amitelle1337 
I think you should be able to not call Keep on the frame-set and only keep the frames you actually need (but not the other way around - you cannot only keep the frameset, since you would run out of color and depth frames)
Ultimately, dispose and keep serve different purposes - 

1. Keep - bypass SDK frame pool to have more frames in memory
2. Dispose - return the frame explicitly to the SDK without waiting for GC

If you are not calling Dispose, frame would only get cleaned-up every couple of seconds when GC is triggered. By that time the SDK may run out of space in internal pool. You could only call Keep and not do dispose, but this way your memory consumption would fluctuate rapidly.

amitelle1337, 2020-09-01:
Thanks for the explanation! :D

closed 2020-09-01 by amitelle1337