realsenseai/librealsense #7202: dark get_color_frame()

linukc, 2020-08-26:
| Camera Model                       | D415 | 
| Operating System & Version |   Ubuntu 18 , Jetpack 4.4 [ L4T 32.4.3]   | 
| Platform                                 | NVIDIA Jetson Nano  |                     
| Language                               |  python | 

### Issue Description
I use simple example from distance_to_object.ipynb (https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks), but something is wrong with frameset.get_color_frame() data. It is very dark photo. Also i noticed that the camera used IR Projector to produce frame.
[image]

MartyG-RealSense, 2020-08-27:
Hi @linukc  I downloaded the **object_detection bag file from the link below so that I could check it in the RealSense Viewer.

http://realsense-hw-public.s3.amazonaws.com/rs-tests/TestData/object_detection.bag

I cannot see any problems with dark RGB in the file.  And because it is recorded data instead of live camera input, it would only respond to changes in the post-processing filter settings and filters are not applied in the **distance_to_object** example.

Could you download the bag file from the link above and drag and drop it into the center window of the RealSense Viewer please to confirm whether the color renders correctly when the bag is played in the Viewer, please?

linukc, 2020-08-27:
@MartyG-RealSense 
[image]

I didn't notice problem with live color stream at another programs (for example jetson example).

MartyG-RealSense, 2020-08-27:
Having a camera connected at the same time as playing the bag should not make a difference to the bag's playback, as it is using recorded data and so the current status of the projector should not make a difference to the bag image.

If you are using that bag file when the camera disconnects, I experienced instability with it myself whilst testing it in the Viewer, such as frequent crash-exits of the Viewer.  Do you still experience camera disconnection if the bag is not being accessed?

linukc, 2020-08-27:
I was able to help myself. I increased Auto-Exposure time to adjust (_for x in range(15):
  pipe.wait_for_frames()_ instead _for x in range(5): pipe.wait_for_frames()_ [The API, line 8]) and color stream works perfect.
Sorry for wasting your time with so stupid problem.

closed 2020-08-27 by linukc

MartyG-RealSense, 2020-08-27:
It's not stupid at all.  We're here to help, and you learned something from it too.  Glad you found a solution!

linukc, 2020-08-27:
@MartyG-RealSense, about disconnect - it not a very big problem, i rarely use Viewer, but i can create another issue and give more details about this strange.

MartyG-RealSense, 2020-08-27:
Okay, please do make a new issue and I will take a look at it on Friday.