realsenseai/librealsense #7204: Capture actions with HW synchronized multiple L515 cameras

MatsumotoJ, 2020-08-27:
Hi,

L515's depth quality is super nice! Thanks for inventing it.

I want to use multiple (3 to 4) L515 cameras with overlapping field of views to capture moving objects.

I checked the recently published [multi-camera white paper](https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup). It is very informative, however, the example system is designed for static scenes but not for dynamic scenes, as it takes at least 93 msec x N cameras for obtaining a single set of frames. 

Is there a way to capture a scene with the full frame rate (30 fps) using the multi-L515 (view-overlapped) configuration?
I guess that the HW triggers in the figure below might be able to slightly delay the timing of laser projection to reduce optical interference. Is it right?
[image]
In my application the USB bandwidth will be ok for 4 cameras, as I use multiple independent USB channels ([like this](https://www.startech.com/Cards-Adapters/USB-3.0/Cards/PCI-Express-USB-3-Card-4-Dedicated-Channels-4-Port~PEXUSB3S44V)).

Your advice would be much appreciated!

RealSenseSupport, 2020-09-11:
Hi @MatsumotoJ 

You are right, the setup within the whitepaper was setup for a static scene body scanning application.  When you say that you want to use the multi camera configuration for motion, can you elaborate a bit more on that?  Maybe give an image or drawing of what you want to setup?  That may help with any advice.

As you read in the whitepaper, the biggest reason for the sync is so that the laser of one camera doesn't interfere with capture of another L515 camera.  Ultimately having one camera active at a time for capture.  As you know there is a delay in capture speed from one camera to another because you're effectively turning off camera from capture while another is capturing and then you need to bring the next camera back up to capture and so on.  

Your statement of "Is there a way to capture a scene with the full frame rate (30 fps) using the multi-L515 (view-overlapped) configuration?".  I'm not sure I understand.  If FOVs will overlap with multiple cameras, then you must control the cameras via syncing in order to turn the other cameras off that will cause IR interference with one camera that is capturing.

Hope this helps.

*If we don’t hear from you in 7 days, this issue will be closed.

MatsumotoJ, 2020-09-12:
Thank @RealSenseSupport,

Here is an example of the setup that I am trying:
[image]

I don't understand the LiDAR mechanism very well, but the simultaneous sampling may be  possible if each camera scan different area of the scene at one time? For example, if L515 scan the scene from top to bottom, this might be possible by  adding slightly different delays (<10 msec) to start scanning to different cameras.

Thank you so much for your support in advance.

MatsumotoJ, 2020-09-12:
PS, 
Because the room size is limited, I can't arrange the cameras to surround the objects. I mean, I can capture only from the front but I want to maximize the 3D information that I get by using the multiple cameras.

MatsumotoJ, 2020-09-18:
I have tested the above idea (starting cameras with different delays) with two L515, and found that it did not reduce interference at all... 

@RealSenseSupport, are you planning to update the SDK or the firmware for the multi-cam motion capture with L515s?

RealSenseSupport, 2020-09-30:
Hi @MatsumotoJ 

The outline in the https://dev.intelrealsense.com/docs/lidar-camera-l515-multi-camera-setup document is what we have with regards to multiple L515 cameras setup.  No plans on modifications to this configuration and SDK.

*If we don’t hear from you in 7 days, this issue will be closed.

MatsumotoJ, 2020-10-01:
@RealSenseSupport, Thanks for the clarification. Then I would try the other camera models for this purpose.

closed 2020-10-01 by MatsumotoJ