realsenseai/librealsense #7206: RPi 4 SDK Version 2.36.0 Available?

swinterbotix, 2020-08-27:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435 | 
| Firmware Version                   | 5.12.06 | 
| Operating System & Version |   Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  5.4                                         | 
| Platform                                 | Raspberry Pi 4b  |
| SDK Version                            |  2.36.0                           | 
| Language                            |  N/A                         | 
| Segment			|  N/A                         | 

### Issue Description

I noticed that there is no librealsense2 2.36.0 version available yet for arm64 type processors (at least for the specs described above). Is this something that's in the works? The most available version currently is 2.35.2.

dorodnic, 2020-08-27:
Hi @swinterbotix 
You should be able to build from source (or cross-compile) any version of the SDK for Arm. At the moment we do provide pre-built releases for Jetson platforms, but not officially for the RPi, yet. We are working to expand the list of platforms with pre-built packages available.

RealSenseCustomerSupport, 2020-08-31:
----------------------------------------------

Hi @swinterbotix Do yo need further help on this? If we don't hear from you in 7 days, this issue will be closed.

swinterbotix, 2020-08-31:
You mentioned that there are no official pre-built releases for the RPi. However, when I type 'apt-cache policy librealsense2' in the pi's terminal, I get a bunch of versions - the most recent being `2.35.2-5ubuntu5~1.gbpd6b9c6` hosted at http://realsense-hw-public.s3.amazonaws.com/Debian/apt-repo bionic/main arm64 Packages. Is this just an unofficial version?

RealSenseSupport, 2020-09-08:
@swinterbotix Yes, please build from source code for RPi as Dorodnic replied. Thanks!

swinterbotix, 2020-09-08:
Alright. Thanks

closed 2020-09-08 by swinterbotix