realsenseai/librealsense #7208: L515 python advanced mode on win10

m10507323, 2020-08-27:
Hi, is it possible to run advance mode in python with L515? Or advance mode is only available in C++?

I follow this sample code but it seems like it can't run with L515.
http://docs.ros.org/kinetic/api/librealsense2/html/python-rs400-advanced-mode-example_8py_source.html

Any help will be appreciated, thanks.

dorodnic, 2020-08-27:
Hi @m10507323 
Advanced Mode is specific to D400 series of cameras. There is no advanced mode API for L515, but also should not be a need for it. 
Advanced device settings are exposed via set/get option APIs.

m10507323, 2020-08-31:
@dorodnic 
Hi, sorry for my lately reply.
Why I ask advanced mode is because I got totally different results from intel viewer and my python program.

Intel viewer result:
[image]
Python result:
[image]

There are two weird problem,
First, in the blue rectangle region, the color information should like the result from intel viewer, different distance has different color, but in my python picture, it just has no difference.
Second, in the white rectangle region, the person's head has same color as the wall in my python result, but the viewer can tell the difference clearly.

And here is my python code,
```python
import pyrealsense2 as rs
import numpy as np
import cv2

pipeline = rs.pipeline()
pipeline.start()
try:
    while True:
        frames = pipeline.wait_for_frames()
        depth = frames.get_depth_frame()
        depth_data = depth.as_frame().get_data()
        np_image = np.asanyarray(depth_data)
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(np_image, alpha=0.03), cv2.COLORMAP_JET)
        cv2.imshow('depth_colormap', depth_colormap)
        cv2.waitKey(1)
finally:
    pipeline.stop()
```
Is there any mistake I make or I miss some preprocessing method in my program?
Thanks for your help.

pitlover, 2020-09-04:
same problems as me. 
Did you solve this problem?

In my problems, I have some difference of depth map between results from my python code and librealsense-viewer. 
my code is below. 

        depth_frame = frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        color_frame = frames.get_color_frame()

        # Validate that both frames are valid
        if not depth_frame or not color_frame:
            print("ERROR : NO FRAME")
            return

        depth_image = np.asanyarray(depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

        # Remove background - Set pixels further than clipping_distance to grey
        grey_color = 153
        depth_image_3d = np.dstack((depth_image,depth_image,depth_image)) #depth image is 1 channel, color is 3 channels
        bg_removed = np.where((depth_image_3d > clipping_distance) | (depth_image_3d <= 0), grey_color, color_image)

        # Render images
        depth_colormap = cv2.applyColorMap(cv2.convertScaleAbs(depth_image, alpha=0.03), cv2.COLORMAP_JET)
        images = np.hstack((bg_removed, depth_colormap))

        result = [[], []]
        result[0].append(depth_image_3d)
        result[1].append(color_image)

        ### DEPTH ARRAY TO IMAGE ###
        cv2.imwrite('{}/DEPTH.png'.format(filename), images)

pitlover, 2020-09-04:
[realsense-viewer]
[image]

[results of my python code]
[image]

Is there any way to solve these problems? Perhaps, changing some options...

m10507323, 2020-09-07:
@pitlover 
Hi, I found this source code on the internet, and it look like similar to intel viewer, maybe you can try this.
I think the problem come from cv2.applyColorMap and cv2.convertScaleAbs, maybe they have different scale method with colorizer.colorize.

```python
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 1024, 768, rs.format.z16, 30)
config.enable_stream(rs.stream.color, 1280, 720, rs.format.bgr8, 30)
profile = pipeline.start(config)
try:
    while True:
        frames = pipeline.wait_for_frames()
        depth = frames.get_depth_frame()
        colorizer = rs.colorizer()
        colorized_depth = np.asanyarray(colorizer.colorize(depth).get_data())
        cv2.imshow('depth_colormap', colorized_depth)
        cv2.waitKey(1)
finally:
    pipeline.stop()
```

pitlover, 2020-09-08:
Thank you! It works well.
Have a nice day~

RealSenseSupport, 2020-09-17:
Hi @m10507323 

Is there anything else needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

m10507323, 2020-09-18:
@RealSenseSupport 
Hi, recently I found out that the image from realsense have more noise and poor resolution than I just got it,
I have already update the firmware, but the image still not good as before,
Can I do anything to reset to default setting?
Thanks.

pitlover, 2020-09-21:
yes! Thank  you

RealSenseSupport, 2020-10-05:
@m10507323 

It seems looking at the IR image in the above images provided that there is a lot of ambient light in the scene that you are capturing.  Are you taking into account the need to have less ambient light in the scene?  Environment definitely has an impact on the depth and noise response seen by the L515 camera.  

https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/
https://www.intelrealsense.com/download/13061/

RealSenseSupport, 2020-10-29:
Is there anything else needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-10-30 by m10507323