realsenseai/librealsense #7209: Unity does not recognize device, but realsense-viewer does

jamjahal, 2020-08-27:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515} | 
| Firmware Version                   | (Open RealSense Viewer --> 01.04.01.02) | 
| Operating System & Version |  MacOS  | 
| Kernel Version (Linux Only)    |  NA                     | 
| Platform                                 | Mac / Unity  |
| SDK Version                            |  {2.36}                          | 
| Language                            |  {C#/opencv/unity }                          | 
| Segment			|  {AR }                          | 

### Issue Description
Hi
As Mentioned above I'm using an l515 lidar camera, which shows up in realsense viewer, but in unity, I keep getting the following error:

ExternalException: rs2_pipeline_start_with_config(pipe:0x7fa321dfe890, config:0x7fa321dfc2c0)
Rethrow as Exception: Couldn't resolve requests
Intel.RealSense.ErrorMarshaler.MarshalNativeToManaged (System.IntPtr pNativeData) (at <58c49cadd7c64bf6b5e89e2d2ff5f02b>:0)
Intel.RealSense.Pipeline.Start (Intel.RealSense.Config cfg) (at <58c49cadd7c64bf6b5e89e2d2ff5f02b>:0)
RsDevice.OnEnable () (at Assets/RealSenseSDK2.0/Scripts/RsDevice.cs:71)

I've seen some comments on similar issues regarding USB versions.  I'm using the supplied cable into a usb3 dongle going into a usb-type c port on a macbook pro.  

I can't get any of the sample scenes to work. They all throw the same error.  As do example scripts like rs-ar-basic, rs-pose, etc.  while examples like depth-quality work.  

Please advise what could be prohibiting my camera from being working.

chrisrogers3d, 2020-08-28:
same for me, i filled an issue a few weeks ago.  supposedly it was fixed in the 2.37 pre-release, but that did not work for me.

ev-mp, 2020-08-28:
@jamjahal hello, the 2.37 version that you're using is [L515-compatible](https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#pre-release-2370), so the issue is either software or setup related.
Inspecting the attached log I see `Couldn't resolve requests` - this can be indicate issue with streaming formats which in turn can be a result  of camera recognized as USB2.
Can you run `rs-enumerate-devices` tool and verify that it reports USB3 connection type?

jamjahal, 2020-08-28:
I see `usb-3.2` next to the camera in realsense-viewer.

`rs-enumerate-device` also shows usb 3.2

[image]

chrisrogers3d, 2020-08-31:
@jamjahal try the new 2.38.1 release from 3 days ago, it worked for me.

RealSenseSupport, 2020-09-11:
Hi @jamjahal 

Were you able to try the latest release?  Are you still seeing an issue with the latest?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-21 by RealSenseSupport