realsenseai/librealsense #7211: Crash when updating L515 Firmware

ve4cib, 2020-08-27:
When I open RealSenseViewer with my L515 connected it displays a prompt saying there's a firmware update available.  Every time I try to install the firmware the application crashes.  This is occurs on Windows 10 & Ubuntu 18.04.

| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
RealSense error calling rs2_stop(sensor:0000022B4206D0D0):
    stop_streaming() failed. UVC device is not streaming!

```

lost-RD, 2020-08-28:
Getting the same issue. Seems like some faulty logic.

ev-mp, 2020-08-28:
@ve4cib , this is an bug that appear in the latest release. We're working on a hotfix for it.
Not that you can upgrade to the new FW using the Viewer's menu
[image]

ev-mp, 2020-08-30:
@ve4cib , @lost-RD  - the hot fix has been merged and retrofitted into the version including both windows and Linux installers
Please check and update.

ev-mp, 2020-09-01:
The issue has  been resolved in the attached PR.
For those who were affected by this bug - please reinstall v2.38.1 distribution packages.

closed 2020-09-01 by ev-mp