realsenseai/librealsense #7212: D455 is unstable in outdoor measurement

gagiter, 2020-08-28:
## Issue Description
I tries the measure a distance between two points in outdoor, but recieve three different values at three samples with more than 10% error. The actual distance is 5m, but the measurements are 4m, 4.32m and 5.16m respectively, as shown in following screenshots.  The default settings are used except the max depth is set to 16m and laser is enabled.

Can i set something to solve this issue? thanks in advance.

| Camera Model                      |                    D455                                   |
|--------------------------------|------------------------------------------- | 
| Operating System                 |             Win (8.1/10)                               |  
| Platform                                |                         PC                                  |
| Max Depth                            |                         16m                               |
| Laser Enabled                        |                         True                               |

[image]
[image]
[image]

Qian Gao

MartyG-RealSense, 2020-08-28:
Hi @gagiter   There are a couple of aspects of your case that I would like to consult with others about before reaching a conclusion.  Thanks for your patience!

MartyG-RealSense, 2020-08-28:
I referred this case to my manager and the issues involved in it are being investigated.

MartyG-RealSense, 2020-09-13:
Adding a note to keep this case open whilst Intel's investigations continue.

MartyG-RealSense, 2020-11-02:
Adding a note to keep this case open for a further time period.

FrankCreen, 2020-11-11:
> ## Issue Description
> I tries the measure a distance between two points in outdoor, but recieve three different values at three samples with more than 10% error. The actual distance is 5m, but the [...]

Hi ！I test the code below with a 75FPS, could you show your result? Thanks very much!

## Test Program
```
// include the librealsense C++ header file
#include <librealsense2/rs.hpp>

#include <opencv2/opencv.hpp>

#include <chrono>

using namespace std;
using namespace cv;

int main()
try
{
    //Contruct a pipeline which abstracts the device
    rs2::pipeline pipe;

    //Create a configuration for configuring the pipeline with a non default profile
    rs2::config cfg;
    //Add desired streams to configuration
    cfg.enable_stream(RS2_STREAM_COLOR, 480, 270, RS2_FORMAT_BGR8, 90);

    //Instruct pipeline to start streaming with the requested configuration
    pipe.start(cfg);

    //Sensor Settings
    auto sensor1 = pipe.get_active_profile().get_device().query_sensors()[1];
    sensor1.set_option(RS2_OPTION_ENABLE_AUTO_EXPOSURE, 1);
    auto sensor2 = pipe.get_active_profile().get_device().query_sensors()[1];
    sensor2.set_option(RS2_OPTION_AUTO_EXPOSURE_PRIORITY, 0);

    // Display in a GUI
    const auto window_name = "Display Video";
    namedWindow(window_name, WINDOW_AUTOSIZE);

    //用于计时
    //show time
    auto start = chrono::system_clock::now();
    int frame = 0;

    while (waitKey(1) < 0 && getWindowProperty(window_name, WND_PROP_AUTOSIZE) >= 0)
    {
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera

        //获取彩色图像帧
        rs2::frame color = data.get_color_frame();

        //保存彩色图像
        Mat image(Size(480, 270), CV_8UC3, (void *)color.get_data(), Mat::AUTO_STEP);

        // Update the window with new data
        imshow(window_name, image);

        frame++;
        //one loop end
        auto end = chrono::system_clock::now();
        auto duration = chrono::duration_cast<chrono::microseconds>(end - start);
        double timecost = double(duration.count()) * chrono::microseconds::period::num / chrono::microseconds::period::den;

        cout << frame / timecost << " FPS" << endl;
    }

    return 0;
}
catch (Exception ex)
{
    cout << "Error!!!" << endl;
    cerr << ex.msg << endl;
}
```

MartyG-RealSense, 2020-11-11:
Hi @FrankCreen    Please post your question as a new issue using the **New Issue** button at the link below and do not post it on multiple D455 cases if the question is unrelated to that case.   Thank you.

https://github.com/IntelRealSense/librealsense/issues/

MartyG-RealSense, 2020-11-21:
Closing this case as the issue is already tracked by Intel on https://github.com/IntelRealSense/librealsense/issues/7155

closed 2020-11-21 by MartyG-RealSense