realsenseai/librealsense #7214: Distance calculation of L515 on the firmware

uma139, 2020-08-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | (Open RealSense Viewer -->1.4.1.2) | 
| Operating System & Version |   Win (10)  | 
| Kernel Version (Linux Only)    |  NA  | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  c++                        | 
| Segment			|               | 

### Issue Description
Hello,

We currently have a L515 LiDAR camera and need information about how the distance is calculated on the camera side(firmware). Which formula/calculation is done?
On the software side, the distance calculation is done using pixel coordinates and depth units. How are the depth units calculated? Could more information be provided regarding the same please. 
Any more information can be provided regarding this question.  Please let us know.

Thank you.

ev-mp, 2020-08-28:
Duplicate of #7146

RealSenseSupport, 2020-09-09:
Hi @uma139 

We don't necessarily get to in depth with regards to providing exactly how depth is calculated by our algorithm.  

We do have some documentation around the camera and the technology that should be helpful but again it won't provide the exacts around our algo calculations.

https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/

https://www.intelrealsense.com/download/13061/

RealSenseSupport, 2020-09-17:
Is there anything else needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-30 by RealSenseSupport