realsenseai/librealsense #7215: L515 drivers not found on Windows 10

SaxonDruce, 2020-08-28:
----------------------------------------------------------------------------------------------------
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Operating System & Version |   Windows 10 - version 2004 | 

### Issue Description

I have an Intel RealSense L515. I have plugged it into a USB 3.0 port on my Windows 10 PC. However it shows up under device manager with this error:

The drivers for this device are not installed. (Code 28)
There are no compatible drivers for this device.
To find a driver for this device, click Update Driver.

[image]

[image]

If I click on 'Update Driver' and then 'Search automatically for drivers', it says 'Windows could not find drivers for your device'.

[image]

I installed all of the latest updates for Windows, in case that was related:

[image]

Is it possible to download the driver from somewhere? The downloads page for the L515 at https://downloadcenter.intel.com/product/201775/Intel-RealSense-LiDAR-Camera-L515 only lists firmware updates, not drivers.

[image]

Thanks.

ev-mp, 2020-08-28:
@SaxonDruce hello, the above snapshot indicates that camera is recognized as USB2.
Currently L515 works with USB3 connection only. Please check the cabling and/or USB port.

SaxonDruce, 2020-08-28:
Hi @ev-mp I tried all of my USB ports again just to be sure - this time it worked fine! Thanks :)

closed 2020-08-28 by SaxonDruce