realsenseai/librealsense #7217: Unable to save the OpenVino identified facial bounding box to a bag file

jack1099, 2020-08-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D415 } | 
| Firmware Version                   | (05.12.06.00) | 
| Operating System & Version |   {ubuntu 18.04}  | 
| Kernel Version (Linux Only)    |  (5.3.0-53-generic)                                         | 
| Platform                                 | HPZ4G4  |
| SDK Version                            |  { 2.36.0 }                          | 
| Language                            |  {Opengl/opencv/C++  }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I am interested to know how to enable the openvino identified facial bounding box to a bag file format

The faces are identified with the bounding box on alive stream. But when I record,and save the recorded steam in .bag file format, the identified facial bounding box is not preserved

As a follow-up on https://github.com/IntelRealSense/librealsense/issues/7143

Can you please point me to line number or code to record and save the identified facial bounding box in .bag file format

dorodnic, 2020-08-30:
BAG files are storing only raw data from the camera, not post-processing and not inference results.

jack1099, 2020-08-30:
@dorodnic Thank you for the response.

 Can you please guide me on how to save the post-processing and inference results to a bag file.

Appreciate your help here.

dorodnic, 2020-08-31:
You cannot save post-processing results to bag file, since as I mentioned, BAG files are storing only raw data from the camera.

jack1099, 2020-08-31:
@dorodnic 

Thank you.

Is there a workaround to save the post processing results in some format.

My requirement is to hide the facial information in the saved recording.

I have come up with a method to hide the facial info. Now I need to save it. Are there any methods to achieve this

[image]

RealSenseSupport, 2020-10-16:
Hi @jack1099 
I think you can create your tool to remove privacy, base on rosbag-inspector.
https://github.com/IntelRealSense/librealsense/tree/master/tools/rosbag-inspector

access the images in the .bag file like this
https://github.com/IntelRealSense/librealsense/blob/master/tools/rosbag-inspector/print_helpers.h#L152-L165

and then, save to another .bag file.

RealSenseSupport, 2020-11-06:
@jack1099  Any other questions about this? Looking forward to your update. Thanks!

RealSenseSupport, 2020-11-30:
@jack1099 Could you please update? Please note that this issue will be closed if we don't get response in 7 days. Thanks!

jack1099, 2020-12-01:
Yes the issue is resolved. Thank you very much.

RealSenseSupport, 2020-12-03:
@jack1099 Thanks for your feedback! Close this issue accordingly.

closed 2020-12-03 by RealSenseSupport