realsenseai/librealsense #7218: Firmware update error in RealSense viewer(2.38.1 RELEASE).

jakemin, 2020-08-28:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE
[image]
[image]

|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
RealSense error calling rs2_stop(sensor:000002D0DB9262C0):
    stop_streaming() failed. UVC device is not streaming!

```

ev-mp, 2020-08-28:
@jakemin, thanks for the report.
Duplicate of #7211

RealSenseCustomerSupport, 2020-09-01:
----------------------------------------------

Hi @jakemin,
thanks for your report
We also able to duplicate this issue on our side
We will looking into this issue

ev-mp, 2020-09-01:
The issue has been resolved (#7211 ).
You'll need to re-download Realsense-Viewer v2.38.1 from [the release's page](https://github.com/IntelRealSense/librealsense/releases/tag/v2.38.1) to proceed.

closed 2020-09-01 by ev-mp