realsenseai/librealsense #7219: Announcement: RealSense SDK version 2.38.1 now available

MartyG-RealSense, 2020-08-28:
Hi everyone,

Version 2.38.1 of the RealSense SDK is now available from the link below.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.38.1 

The full details of API and SDK changes can be read at the links below.

**API**
https://github.com/IntelRealSense/librealsense/wiki/API-Changes#version-2381 

**SDK**
https://github.com/IntelRealSense/librealsense/wiki/Release-Notes#release-2381

huoxinr1998, 2020-08-28:
unfortunately，THE Link"https://github.com/IntelRealSense/librealsense/releases/tag/v2.38.1 " is 404 error. Please update it.

TheBricktop, 2020-08-28:
Great news, I like the fixes for l515. Still the lack of support for t265 on Android is quite a pain for me and my project. Anyways keep up the good work. I'd really like to use Android because of better 3d engines support than other arm Linux distros, and native vr support.

MartyG-RealSense, 2020-08-28:
Thanks so much for the feedback @TheBricktop    :)

huoxinr1998, 2020-08-30:
When will the LabVIEW examples support L515 devices？My projects urgently need it

MartyG-RealSense, 2020-08-30:
Hi  @huoxinr1998  Can you create a new case to ask this question in please by visiting the front page of this GitHub forum at the link below and clicking the **New Issue** button.  This will bring it to the attention of RealSense team members who handle L515 questions.   Thanks!

https://github.com/IntelRealSense/librealsense/issues

chrisrogers3d, 2020-08-31:
Thank you! I can now use the 515 in Unity.  I can use a 435, 455, and 515 at the same time(!)  there seems to be an issue with mixing up rgb streams, if i cant figure it out, ill make a test case.

MartyG-RealSense, 2020-08-31:
Great to hear @chrisrogers3d  - if you do find that you need assistance, feel free to create a case on the forum to ask about it. Thanks!

TheBricktop, 2020-09-02:
I don't know if this is the proper place to address this issue but ive asked in other places and didnt hear any answear and this is pretty vital for my project:
There is a merge request for changes that enable t265 support on android but they are stalling on review  could anyone with proper authority look on it and tell us if this is viable for merging or not and why? @MartyG-RealSense  @smartynenko  @dorodnic 
Enable android support for t265 #6011

MartyG-RealSense, 2020-09-02:
@TheBricktop   I'm not involved in T265 related issues as other Intel RealSense team members are responsible for that camera.

TheBricktop, 2020-09-11:
@MartyG-RealSense sorry for asking i was thinking that You might have some informations about it.  There was no direct answer  about the status of android support for t265, neither about the stalled merge request in #6011. I hope it is not because some NDA or other legal issues.

MartyG-RealSense, 2020-09-11:
@TheBricktop   I will refer your question to @RealSenseSupport for a response.

TheBricktop, 2020-09-11:
Thank You @MartyG-RealSense .

closed 2020-09-15 by MartyG-RealSense