realsenseai/librealsense #7220: Can't click on realsense device in Add Source menu in Realsense Viewer

t-kole-yipp, 2020-08-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D400 series } | 
| Firmware Version                   | 2.38.1.2223 | 
| Operating System & Version |  Win 10 | 

It happens fairly often that I can't click on a device in Realsense Viewer to add it as a source.

- I'm using the latest version
- The device is plugged in
- The device is listed in the "add source" list
- The device even works in my own application, using the Unity package.
- I can click on it, but nothing happens
- There are no error or warning messages, not when starting the application nor when clicking on the item.
- This applies to only some devices in the list. Others do work.
- It will happen with both D435 and D435i's.

[image]

MartyG-RealSense, 2020-08-28:
Hi @t-kole-yipp   There was a similar case in May 2020 with selection of multiple cameras in the RealSense Viewer.  Initially the RealSense user found that they had problems with 20 cameras listed in the Viewer, though they also found that they had the problem with just 2 cameras.

https://github.com/IntelRealSense/librealsense/issues/6440

At the time it was decided not to submit an official bug report to Intel as there appeared to be nobody else who was affected by the problem.  It may be worth checking with the RealSense user involved in the above case.

Hi @MojamojaK  Are you still affected please by the multiple camera selection problem in the Viewer that you reported in May 2020, please?

t-kole-yipp, 2020-08-28:
I wanted to add that this is a frequent problem here, and it happens on multiple PC's (all with Windows 10).
It's been around for a few versions I believe, but it's hard to replicate consistently.
Sometimes a reboot helps, sometimes it doesn't.
Sometimes plugging one out and in helps, sometimes it doesn't.

MartyG-RealSense, 2020-08-28:
@t-kole-yipp   Thanks very much for the added information.  As it appears that you can replicate the circumstances of the earlier case, it may be worth considering filing an official bug report so that RealSense team members can take a look at the issue and see if they can replicate it themselves.

Given that this seems to be an issue that has been occurring over the long term, let's see what other users and RealSense team members such as @dorodnic  contribute to this discussion.

MojamojaK, 2020-08-28:
Yep, still the same thing for me

MartyG-RealSense, 2020-08-29:
Thanks very much @MojamojaK for the confirmation that you are still affected.

I'll wait another day to see if anybody else sees this case and reports similar symptoms.

MartyG-RealSense, 2020-09-09:
Given that more than one person has reported the issue described in this case over a long term period, I will file a bug report with Intel so that the RealSense team can investigate.   Thanks so much for the patience of everybody involved in this case.

MartyG-RealSense, 2022-02-13:
Case closed due to no further reports of this issue occurring since 2020.

closed 2022-02-13 by MartyG-RealSense