realsenseai/librealsense #7222: How can I get gyro and accel values from L515 at MATLAB?

jrh3503, 2020-08-28:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | Development 1.4.1.2 | 
| Operating System & Version |   Win 10  |  
| Platform                                 | PC  |
| SDK Version                            |  SDK v2.36.0                          | 
| Language                            |  MATLAB                          | 
| Segment			|  Robot}                          | 

### Issue Description
I am trying to implement SLAM in MATLAB using L515. So I try to get the gyro and accel values from L515, but I can't get them.
First of all, it says that get_pose_data cannot be performed when the code below is written and executed (I don't understand why it is made into video_frame class, not pose_frame).

%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%

del = exist ('pipe','class');
if ( ( del ~= 0) )
pipe.stop();
pipe.delete;
clear('pipe');
end

clc;
clear;

width_depth = 640;
height_depth = 480;
width_infrared = 640;
height_infrared = 480;
width_rgb = 1280;
height_rgb = 720;
fps = 30;

pipe = realsense.pipeline();
cfg = realsense.config();
cfg.enable_stream(realsense.stream.color,width_rgb,height_rgb,realsense.format.rgb8,fps);
cfg.enable_stream(realsense.stream.depth,width_depth,height_depth,realsense.format.z16,fps);
cfg.enable_stream(realsense.stream.infrared,width_infrared,height_infrared,realsense.format.y8,fps);
cfg.enable_stream(realsense.stream.gyro,realsense.format.motion_xyz32f);
cfg.enable_stream(realsense.stream.accel,realsense.format.motion_xyz32f);

profile = pipe.start(cfg);

for i = 1:5 %% 1:20
fs = pipe.wait_for_frames();
end

while(1)
%Wait for the frames
fs = pipe.wait_for_frames();
%Select color frame
color = fs.get_color_frame();
infrared = fs.get_infrared_frame();

% Select depth frame
depth = fs.get_depth_frame();
%Get actual data and convert into a format imshow can use
data_color = color.get_data();
data_infrared = infrared.get_data();

pos= fs.get_pose_frame();
pos_data = pos.get_pose_data();

% %(Color data arrives as [R, G, B, R, G, B, ...] vector)
% %img(:,:,1) = red,img(:,:,2) = green,img(:,:,3) = blue
%img = permute(reshape(data_color,[3,color.get_width(),color.get_height()]),[3 2 1]); %color frame 
img = permute(reshape(data_infrared,[infrared.get_width(),infrared.get_height()]),[2 1]); %infrared frame

%Display the image
imshow(img);

end

%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%

I want you to tell me why the problem occurs above and how we can solve it." (Is there a problem on SDK's side, or is my code a problem?)
The best thing is, as mentioned above, I would like you to tell me about examples where you can get gyro and accel values from matlab, or things you can refer to.

kind regards

RealSenseSupport, 2020-09-12:
Hi @jrh3503 

Our Matlab support is delayed for L515.  It will take us some time to provide Matlab updates.

RealSenseSupport, 2021-01-12:
Hi @jrh3503 

Not sure if you saw this PR that could help here:  https://github.com/IntelRealSense/librealsense/pull/7880

Please have a look and let us know if it works for you.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2021-01-28 by RealSenseSupport