realsenseai/librealsense #7223: no device connected after run realsense-viewer

linukc, 2020-08-28:
| Camera Model                       | D415  | 
| Firmware Version                   | 05.12.05.00 | 
| Operating System & Version |   Ubuntu 18 , Jetpack 4.4 [ L4T 32.4.3]  |                                     
| Platform                                 | NVIDIA Jetson Nano  |
| Language                               |  python |                       

### Issue Description
 Step to reproduce:

1. Connect camera.

3. Run simple script ( https://gist.github.com/linukc/42bde14c60373e822cb64f29e5a30ad0 ). Evrething is ok.

2. Run realsense-viewer.

3. Close realsense-viewer.

4. Run script again. _**Error: no device connected.**_ So i have to reconnect the camera.

[image]
[image]

MartyG-RealSense, 2020-08-28:
Hi @linukc   Which librealsense version are you using, please?

linukc, 2020-08-29:
@MartyG-RealSense v2.35.2

MartyG-RealSense, 2020-08-29:
You could try putting a **hardware_reset** camera reset instruction near the start of your script.  A piece of Python code for doing so is in the link below.

https://github.com/IntelRealSense/librealsense/issues/3329#issuecomment-475293475

And here is a slightly more advanced example that resets a camera with a particular serial number:

https://github.com/IntelRealSense/librealsense/issues/5428#issuecomment-564482167

linukc, 2020-08-29:
@MartyG-RealSense , thanks.

closed 2020-08-29 by linukc