realsenseai/librealsense #7224: L515 not detected in realsense-viewer v2.36.0 with OS Ubuntu 18.04

gauvainrobert, 2020-08-28:
Hello,

I bought an **L515** recently. The camera is detected by my **OS Ubuntu 18.04** but when I am running **realsense-viewer** (v 2.36.0) the camera is not detected. Before this camera, **I was using a D435 camera and it was working perfectly**.

**realsense-viewer** is returning this at start:

>  28/08 18:34:47,382 ERROR [139676152788736] (types.h:308) xioctl(VIDIOC_S_CTRL) failed Last Error: Invalid or incomplete multibyte or wide character
>  28/08 18:34:47,526 WARNING [139676152788736] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
>  28/08 18:34:47,527 ERROR [139676152788736] (types.h:308) xioctl(VIDIOC_G_CTRL) failed Last Error: Invalid argument
>  28/08 18:34:47,527 WARNING [139676152788736] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Enable / disable auto-exposure

The camera is detected when using **lsusb**

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 |
| Operating System & Version |   {Linux (Ubuntu 18.04)  | 
| Kernel Version (Linux Only)    |  5.4.0-42-generic                                        | 
| Platform                                 | PC |
| SDK Version                            |  { 2.0 }                          |

ev-mp, 2020-08-28:
@gauvainrobert, the report is not related to L515 but the embedded/web cam that is attached to host as well ([6784](https://github.com/IntelRealSense/librealsense/issues/6784#issuecomment-662108300))
>Exception was thrown when inspecting **Raw RGB Camera** 

As for L515 not being listed - please share the kernel log when connecting camera  `dmesg -wT` . That line is a hint for improper enumeration or missing UVC driver
>The camera is detected when using lsusb

gauvainrobert, 2020-08-28:
@ev-mp Thank you for the reply and the first information, concerning the command `dmesg -wT`, the result is:

> [ven. août 28 19:26:27 2020] usb 1-2: new high-speed USB device number 7 using xhci_hcd
> [ven. août 28 19:26:27 2020] usb 1-2: New USB device found, idVendor=8086, idProduct=0b64, bcdDevice=10.41
> [ven. août 28 19:26:27 2020] usb 1-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
> [ven. août 28 19:26:27 2020] usb 1-2: Product: Intel(R) RealSense(TM) 515
> [ven. août 28 19:26:27 2020] usb 1-2: Manufacturer: Intel(R) RealSense(TM) Camera
> [ven. août 28 19:26:27 2020] usb 1-2: SerialNumber: 00000000F0090415

ev-mp, 2020-08-28:
@gauvainrobert , thank you for the prompt reply.
As suspected the issue is with incorrect enumeration:
>[ven. août 28 19:26:27 2020] usb 1-2: new **high-speed USB** device number 7 using xhci_hcd

The camera is recognized as USB2 but requires USB3 connection. Please try to check cable/switch port till a `new super-speed USB device` is reported.

closed 2020-08-28 by gauvainrobert

Ryanf55, 2024-09-07:
I get the following dmesg output. It recognizes as a `super-speed` device, but does not show up in the realsense viewer.
```
[Sun Sep  8 17:03:54 2024] usb 2-2: new SuperSpeed USB device number 4 using xhci_hcd
[Sun Sep  8 17:03:54 2024] usb 2-2: New USB device found, idVendor=8086, idProduct=0b64, bcdDevice=10.3f
[Sun Sep  8 17:03:54 2024] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[Sun Sep  8 17:03:54 2024] usb 2-2: Product: Intel(R) RealSense(TM) 515
[Sun Sep  8 17:03:54 2024] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Camera
[Sun Sep  8 17:03:54 2024] usb 2-2: SerialNumber: 00000000F0172280
[Sun Sep  8 17:03:54 2024] usb 2-2: Found UVC 1.50 device Intel(R) RealSense(TM) 515 (8086:0b64)
[Sun Sep  8 17:03:54 2024] usb 2-2: Found UVC 1.50 device Intel(R) RealSense(TM) 515 (8086:0b64)
```

```bash
$ lsusb | grep 515
Bus 002 Device 005: ID 8086:0b64 Intel Corp. Intel(R) RealSense(TM) 515
```

OS: Ubuntu 22

Here's a screenshot of what the realsense viewer show, with logs.
[image]

MartyG-RealSense, 2024-09-07:
Hi @Ryanf55  Your screensshot shows that you are using RealSense Viewer version 2.55.1.  Support for the L515 camera model ws removed in version 2.54.1 onwards of the RealSense SDK (librealsense) and the RealSense Viewer.  Your L515 should be able to be accessed in the RealSense Viewer if you install the **2.51.1** version of the SDK.

https://github.com/IntelRealSense/librealsense/releases/tag/v2.51.1

Ryanf55, 2024-09-07:
Thanks. Perhaps the getting started guide could be updated?
`intelrealsense.com/get-started` on Ubuntu does not mention the debians are incompatible. 
It brought me to the install here:
https://github.com/IntelRealSense/librealsense/blob/development/doc/distribution_linux.md
Which also does not mention L515 is no longer supported.

Ryanf55, 2024-09-07:
Ah, you forgot to mention. You don't support Ubuntu 22 on v2.51.1 per this [comment](https://github.com/IntelRealSense/librealsense/issues/10439#issuecomment-1357265581)

It doesn't compile without modifications
```
In file included from /home/ryan.friedman/Dev/librealsense/src/libusb/request-libusb.h:6,
                 from /home/ryan.friedman/Dev/librealsense/src/libusb/request-libusb.cpp:4:
/home/ryan.friedman/Dev/librealsense/src/libusb/libusb.h:10:82: error: extended character ‘ is not valid in an identifier
   10 | /usr/include/libusb-1.0/libusb.h:736:4: warning: ISO C++ forbids zero-size array ‘dev_capability_data’ [-Wpedantic]
```
If I delete that code, it compiles.

MartyG-RealSense, 2024-09-08:
Models that are no longer supported due to retirement are not listed in the SDK documentation.  

In regard to L515 though there have been cases of users attempting to install the recent SDK and finding that their camera was not detected.  Intel are planning to address this by adding a mechanism to the SDK that informs the user that L515 is not supported if they attempt to use that model wirh the SDK.

Thanks very much for sharing your code deletion method!

Ryanf55, 2024-09-08:
> Models that are no longer supported due to retirement are not listed in the SDK documentation.
>
> In regard to L515 though there have been cases of users attempting to install the recent SDK and [...]

That would be a much nicer user experience, perhaps it could suggest the last supported version to try.

> Thanks very much for sharing your code deletion method!

Would you be open to a patch to the last known release to replace the backticks with single quotes so it can compile?

MartyG-RealSense, 2024-09-08:
Thanks very much for the offer of a patch to 2.51.1.  Previous versions of the SDK are not retroactively updated though, so such a patch could not be implemented into a past version.

It is very helpful though that you have posted your edit here, as it will be findable by other L515 users.