realsenseai/librealsense #7227: Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:

wykasz, 2020-08-28:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Unhandled exception escaping from a worker thread!
Error type: Access Violation!
```

ev-mp, 2020-08-29:
@wykasz , can you provide log/screenshots and describe the flow that triggered that error.
Additionally please attach the camera info generated with `rs-enumerate-devices -c`

wykasz, 2020-08-29:
Hi
It happend during the first start of the Intel.RealSense.Viewer.exe app.
The device was connected to usb 2.0 port of lenovo e490 notebook with Windows 10 pro.
After that i've tested it under usb.30 and it happend again.
Write now the device works with usb 2.0 but connetcing it to usb 3.0 triggers that error all the time, making the device unusable.

I've attached screenshots below.
[image]
[image]

I'll be happy to send you logs, if You just could tell me where i can find them.

rs-enumerate-devices tool unser usb 3.0 fails
 ./rs-enumerate-devices -c

>  29/08 20:59:48,468 ERROR [20300] (types.h:313) hr returned: HResult 0x800706be:
>  "Zdalne wywo┼éanie procedury nie powiod┼éo si─Ö."
> Could not create device - hr returned: HResult 0x800706be: "Zdalne wywo┼éanie pr
> ocedury nie powiod┼éo si─Ö." . Check SDK logs for details
> 

Under usb 2.0 it returns that

[rs-enumerate-devices_output.txt](https://github.com/IntelRealSense/librealsense/files/5145883/rs-enumerate-devices_output.txt)

Regards

RealSenseSupport, 2020-08-31:
@wykasz Which RealSense camera you are using?

wykasz, 2020-08-31:
Sorry, D435i

ev-mp, 2020-09-01:
@wykasz , the attached log file provides details enumeration with USB3.
And in the first attached screenshot, the connection type is shown as USB2 (in the upper right) - Can you clarify whether the situation occurs under USB2 or USB3?

Additionally, from the log I can see that the Firmware version used 5.11.1.100 is rather old (Jan 2019).
Can you upgrade to the latest recommended Firmware using Realsense-Viewer and then try to re-enumerate it with USB2 port ?
It could have been the case with USB2 support for D435i in that particular version.

ev-mp, 2020-09-06:
@wykasz, did you manage to upgrade the FW to the (minimal) recommended version ?

RealSenseCustomerSupport, 2020-09-24:
----------------------------------------------

Hi @wykasz, Do you need further help on this? If we don't hear from you in 7 days, this issue will be closed.

closed 2020-10-05 by RealSenseCustomerSupport