realsenseai/librealsense #7228: distorted 3d image

netanelfar, 2020-08-29:
| | |
|---|---|
|**librealsense**|2.36.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense D435|
|**Serial Number**|740112070993|
|**Firmware Version**|05.12.06.00|
|**Advanced Mode**|YES|
|**Camera Locked**|YES|
|**Usb Type Descriptor**|3.2|
|**Product Line**|D400|
|**Asic Serial Number**|828313020569|
|**Firmware Update Id**|828313020569|

Please provide a description of the problem
since the camera got a mild hit the input is a follow:

[image]

[image]

is it a hardware problem? did somthing broke or do i need to recalibrate?

MartyG-RealSense, 2020-08-29:
Hi @netanelfar   A 400 Series camera can become mis-calibrated if it receives a physical shock such as a hard knock or a drop on the ground.  It can be re-calibrated with software such as the **Dynamic Calibrator** tool.

https://downloadcenter.intel.com/download/29618/Intel-RealSense-D400-Series-Dynamic-Calibration-Tool

Aside from the black areas, there does not seem to be serious problems with the image though.   It may therefore be sufficient to first try using the **Self Calibration** function that is built into the RealSense Viewer to inprove depth quality.  You can launch this tool by going to the **More** option at the top of the Viewer's side-panel and selecting the **On-Chip Calibration** option from its menu.

netanelfar, 2020-08-29:
@MartyG-RealSense thanks,
after using On-Chip Calibration option and running my test code i got this:

[image]

as you can see the image is distorted even more.
is should i try Dynamic Calibrator?

MartyG-RealSense, 2020-08-29:
The blue '3D' option in the top corner of the Viewer indicates that you are in 3D mode.  This is a type of depth sensing scan made from a cloud of points.  Also, in the last of the images above you have the depth stream disabled, leaving just a broken looking RGB image.  It is typical for the RGB to look incomplete in this mode.

To get a normal unbroken RGB image, you should click the '2D' option.

MartyG-RealSense, 2020-09-09:
Hi @netanelfar    Do you require further assistance with this case, please?   Thanks!

netanelfar, 2020-09-10:
hi @MartyG-RealSense the problem still exist i tried to recalibrate but hr edges are still distorted, is there anything else i can do to try to fix it?

MartyG-RealSense, 2020-09-10:
I ran extensive further tests in 3D point cloud mode with a D435 with color mapped to depth and was able to recreate the characteristics of your scene.  The only real way I found to make a further improvement, such as reducing the corner distortion, was to move the camera further away from the observed scene.

netanelfar, 2020-09-10:
@MartyG-RealSense . thanks for your help, ill close it for now

closed 2020-09-10 by netanelfar