realsenseai/librealsense #7229: CUDA 11 support?

jiapei100, 2020-08-29:
### Building environment

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | N/A | 
| Firmware Version                   | N/A | 
| Operating System & Version |   Linux (Ubuntu 20.04)  | 
| Kernel Version (Linux Only)    |  5.4.0-42-generic          | 
| Platform                                 | PC  |
| SDK Version                            | 2.38.1                      | 
| Language                            |  C++                          | 
| Segment			|  Everything                          | 

### Issue Description

It looks like librealsense doesn't support CUDA 11 yet? ???

```console
➜  ~ nvcc --version
nvcc: NVIDIA (R) Cuda compiler driver
Copyright (c) 2005-2020 NVIDIA Corporation
Built on Wed_Jul_22_19:09:09_PDT_2020
Cuda compilation tools, release 11.0, V11.0.221
Build cuda_11.0_bu.TC445_37.28845127_0
```

```console
CMake Error at /usr/share/cmake-3.16/Modules/CMakeTestCUDACompiler.cmake:46 (message):
   The CUDA compiler

     "/usr/local/cuda/bin/nvcc"

   is not able to compile a simple test program.

   It fails with the following output:

     Change Dir: ......librealsense/build/CMakeFiles/CMakeTmp

     Run Build Command(s):/usr/bin/make cmTC_336f7/fast && /usr/bin/make -f CMakeFiles/cmTC_336f7.dir/build.make CMakeFiles/cmTC_336f7.dir/build
     make[1]: Entering directory '......librealsense/build/CMakeFiles/CMakeTmp'
     Building CUDA object CMakeFiles/cmTC_336f7.dir/main.cu.o
     /usr/local/cuda/bin/nvcc     -x cu -c ......librealsense/build/CMakeFiles/CMakeTmp/main.cu -o CMakeFiles/cmTC_336f7.dir/main.cu.o
     /usr/include/c++/10/type_traits(1384): error: type name is not allowed

     /usr/include/c++/10/type_traits(1384): error: type name is not allowed

     /usr/include/c++/10/type_traits(1384): error: identifier "__is_same_as" is undefined

     3 errors detected in the compilation of "......librealsense/build/CMakeFiles/CMakeTmp/main.cu".
     make[1]: *** [CMakeFiles/cmTC_336f7.dir/build.make:66: CMakeFiles/cmTC_336f7.dir/main.cu.o] Error 1
     make[1]: Leaving directory '......librealsense/build/CMakeFiles/CMakeTmp'
     make: *** [Makefile:121: cmTC_336f7/fast] Error 2

   CMake will not be able to correctly generate this project.
 Call Stack (most recent call first):
   CMake/cuda_config.cmake:3 (project)
   CMake/global_config.cmake:65 (include)
   CMakeLists.txt:44 (global_set_flags)
```

Cheers

ev-mp, 2020-09-01:
@jiapei100 , the error is due to C++/NVCC compiler compatibility issue.
There are several similar issues being tracked, with some progress reported - [link1](https://xpra.org/trac/ticket/1800?cversion=0&cnum_hist=4), [link2](https://gist.github.com/ax3l/9489132).

Please verify the GCC/Clang version you use against NVidia's compatibility matrix - https://docs.nvidia.com/cuda/cuda-installation-guide-linux/index.html#axzz3x7mwvZrG

RealSenseSupport, 2020-09-03:
Please kindly check the compatibility matrix as suggested and let us know if further help is needed in regard to this issue. 
Otherwise, this ticket will be closed in 7 days. Thank you.

closed 2020-09-11 by RealSenseSupport