realsenseai/librealsense #7230: L515 Segfault with Python

JeremyBYU, 2020-08-29:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515  | 
| Firmware Version                   | 01.05.00.00 | 
| Operating System & Version |   Linux | 
| Kernel Version (Linux Only)    |  5.3                                         | 
| Platform                                 | PC X64  |
| SDK Version                            |  2.38                         | 
| Language                            |  Python                         | 
| Segment			|  Robot                          | 

### Issue Description
L515 will segfault after pipeline is started. This **only** occurs after a `context.devices()` API is called **and** the subsequent retrieval of any device from the list. This error does not occur with D4XX Cameras. Below is reproducable script.

```python
import pyrealsense2 as rs
import sys

ctx = rs.context()
devices = ctx.query_devices()
# Uncomment below to see eventual segfault with L515, no error with D4XX
# dev = devices[0] # <-------- This is the specifc line that causes an error
# dev_name = dev.get_info(rs.camera_info.name)
# print("Found {}".format(dev_name))

# Configure stream
pipeline = rs.pipeline()
config = rs.config()
config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)

# Start streaming
# This will start a thread that will eventually segfault with L515, no error with D4XX
pipeline.start(config)
count = 0

while True:
    try:
        frames = pipeline.wait_for_frames(timeout_ms=1000)
        print(frames)
    except:
        print("Frames didn't arrive in time")
        count += 1
        if count > 2:
            sys.exit()

```

maloel, 2020-08-31:
Hi @JeremyBYU 
We have not been able to reproduce your issue. Two things come to mind: either it's the kernel version that you're using, or there might be some other webcam connected to your computer that be getting in the way.

Can you try to disconnect any other webcam and see if this reproduces?

Can you supply us with a librealsense log? You can add a line:
`rs.log_to_file( 'lrs.log' )`
Place this right after the import.

JeremyBYU, 2020-08-31:
1. There are no other additional webcams on this system (desktop system).
2. Here are the details on my kernel version - `Linux cerberus 5.3.0-7648-generic #41~1586790036~18.04~600aeb5-Ubuntu SMP Mon Apr 13 17:47:15 UTC  x86_64 x86_64 x86_64 GNU/Linux
`

Since you were not able to reproduce I did some more work verifying the issue. I remembered that my RealSense version was actually from the development branch around 2.38. Specifically this commit was the last one: 

```
commit 7d75e327b8d009e27ad264d2fc32c1ceb68936df
Merge: 4cc9820b2 f31bc43f5
Author: ev-mp <ev_mp@yahoo.com>
Date:   Tue Aug 4 19:38:52 2020 +0300

    Merge pull request #6933 from icarpis/RelComPtr
    
    Release CComPtr cause to access violation

``` 

I can consistently reproduce the segmentation fault for L515 on this commit. However I did a `git pull` and updated to the latest development branch and now the segmentation fault is gone. I am now specifically on:

```
commit d421050f591f90e16dbed9b2efe21e89da54e294 (HEAD -> development, origin/development)
Merge: e77bd2f4c e87364e97
Author: Sergey Dorodnicov <sergey.dorodnicov@intel.com>
Date:   Mon Aug 31 17:41:32 2020 +0300

    Merge pull request #7213 from libdavid/development
    
    Fixing a wrong argument in "CreateEvent" call

```

I am sorry about the false alarm. I will remember to pull from the development branch before reporting a bug in the future (or only file bugs which are from a recent release).

closed 2020-08-31 by JeremyBYU