realsenseai/librealsense #7231: realsense-viewer crash

viisautta, 2020-08-30:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | 05.12.07.100, USB type: 3.2 | 
| Operating System & Version |   Fedora 32  | 
| Kernel Version (Linux Only)    | 5.7.17-200.fc32.x86_64 | 
| Platform                                 | lapbook |
| SDK Version                            | 2.38.1  | 
| Language                            |  C++ | 
| Segment			|  others | 

### Issue Description
$ ./realsense-viewer 
 30/08 18:10:04,964 ERROR [140267030038272] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 30/08 18:10:04,965 WARNING [140267348993792] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Enable / disable backlight compensation
 30/08 18:10:04,965 WARNING [140267348993792] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image brightness
 30/08 18:10:04,965 WARNING [140267348993792] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image contrast
 30/08 18:10:04,965 WARNING [140267348993792] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 30/08 18:10:04,965 WARNING [140267348993792] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image gamma setting
 30/08 18:10:04,965 WARNING [140267348993792] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image hue
 30/08 18:10:04,965 WARNING [140267348993792] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image saturation setting
Segmentation fault (core dumped)

RealSenseSupport, 2020-09-03:
@viisautta Could you please share what build options you set? Looking forward to your update. Thanks!

viisautta, 2020-09-04:
ccmake shows as below:

 ANDROID_USB_HOST_UVC             OFF
 APRILTAG_INC                     /usr/local/apriltag/include/apriltag       
 APRILTAG_LIB                     /usr/local/apriltag/lib64/libapriltag.so
 BUILD_CSHARP_BINDINGS            OFF
 BUILD_CV_EXAMPLES                ON
 BUILD_CV_KINFU_EXAMPLE           ON
 BUILD_DLIB_EXAMPLES              OFF
 BUILD_EASYLOGGINGPP              ON
 BUILD_EXAMPLES                   ON
 BUILD_GLSL_EXTENSIONS            ON
 BUILD_GRAPHICAL_EXAMPLES         ON
 BUILD_INTERNAL_UNIT_TESTS        OFF
 BUILD_MATLAB_BINDINGS            OFF
 BUILD_NETWORK_DEVICE             OFF
 BUILD_NODEJS_BINDINGS            OFF
 BUILD_OPENNI2_BINDINGS           OFF
 BUILD_OPENVINO_EXAMPLES          OFF
 BUILD_PCL_EXAMPLES               ON
 BUILD_PYTHON_BINDINGS            OFF
 BUILD_PYTHON_DOCS                OFF
 BUILD_SHARED_LIBS                ON
 BUILD_UNITY_BINDINGS             OFF
 BUILD_UNIT_TESTS                 OFF
 BUILD_WITH_CUDA                  OFF
 BUILD_WITH_OPENMP                OFF
 BUILD_WITH_STATIC_CRT            ON
 BUILD_WITH_TM2                   OFF
 CCACHE_FOUND                     /usr/bin/ccache
 CHECK_FOR_UPDATES                OFF
 CMAKE_BUILD_TYPE                 Release
 CMAKE_INSTALL_PREFIX             /usr/local/librealsense-2.38.1
 ENABLE_CCACHE                    ON
 ENABLE_L500_DEPTH_INVALIDATION   OFF
 ENABLE_ZERO_COPY                 ON
 ENFORCE_METADATA                 ON
 FORCE_LIBUVC                     OFF
 FORCE_RSUSB_BACKEND              ON
 FORCE_WINUSB_UVC                 OFF
 GLEW_DIR                         GLEW_DIR-NOTFOUND
 HWM_OVER_XU                      ON
 IMPORT_DEPTH_CAM_FW              OFF
 LIBUSB_1_INCLUDE_DIR             /usr/include
 LIBUSB_1_LIBRARY                 /usr/lib64/libusb-1.0.so
 LIBUSB_INC                       /usr/include/libusb-1.0
 LIBUSB_LIB                       /usr/lib64/libusb-1.0.so
 OPENNI2_INCLUDE_DIR              /home/daqian/Public/Devel/OpenNI-Linux-x64-2.2/Include
 OPENNI2_LIBRARY                  /home/daqian/Public/Devel/OpenNI-Linux-x64-2.2/Redist/libOpenNI2.so
 OPENNI_INCLUDE_DIR               /usr/include/ni
 OPENNI_LIBRARY                   /usr/lib64/libOpenNI.so
 OpenCV_DIR                       /usr/local/opencv-4.4.0/lib64/cmake/opencv4
 PCL_DIR                          /usr/local/pcl-1.11.0/share/pcl-1.11    
 QHULL_LIBRARY                    /usr/lib64/libqhull_p.so
 QHULL_LIBRARY_DEBUG              /usr/lib64/libqhull_p.so
 Qt5Core_DIR                      /usr/lib64/cmake/Qt5Core
 Qt5Gui_DIR                       /usr/lib64/cmake/Qt5Gui
 Qt5Widgets_DIR                   /usr/lib64/cmake/Qt5Widgets
 USB_10_INCLUDE_DIR               /usr/include
 USB_10_LIBRARY                   /usr/lib64/libusb-1.0.so
 VTK_DIR                          /usr/lib64/cmake/vtk
 flann_DIR                        /usr/local/flann/lib64/cmake/flann
 glfw3_DIR                        /usr/lib64/cmake/glfw3

viisautta, 2020-09-08:
@RealSenseSupport pls see above:)

RealSenseSupport, 2020-09-08:
@viisautta Did you get any issue when run other SDK examples? Such as rs-capture or rs-depth-quality?

viisautta, 2020-09-10:
@RealSenseSupport 
At present, rs-capture or rs-depth-quality is available without crashes.

RealSenseSupport, 2020-09-14:
@viisautta  We tried RealSenseViewer on one setup with Fedora 32 but didn't reproduce the crash issue. Another thing to check with you that the crash on your side happens just after execute RealSenseViewer or the GUI can be launched and crash happens after the streaming turned on? If it happens after start streaming, what streaming and configuration(resolution/fps) do you use when get the issue? Looking forward to your update. Thanks!

viisautta, 2020-09-17:
I only tried this bin file some days ago without too much attention to resolution/fps and etc..
Now I don't know why this GUI is displaying, and other bin files like rs-gl, rs-align ... work well.
but there are some errors and warnings as below:
$ ./realsense-viewer 
 17/09 22:04:43,622 ERROR [139686152496896] (handle-libusb.h:51) failed to open usb interface: 0, error: RS2_USB_STATUS_ACCESS
 17/09 22:04:43,625 WARNING [139686479652608] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Enable / disable backlight compensation
 17/09 22:04:43,625 WARNING [139686479652608] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image brightness
 17/09 22:04:43,625 WARNING [139686479652608] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image contrast
 17/09 22:04:43,625 WARNING [139686479652608] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls exposure time of color camera. Setting any value will disable auto exposure
 17/09 22:04:43,625 WARNING [139686479652608] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image gamma setting
 17/09 22:04:43,625 WARNING [139686479652608] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image hue
 17/09 22:04:43,625 WARNING [139686479652608] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image saturation setting
 17/09 22:04:43,625 WARNING [139686479652608] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property UVC image sharpness setting
 17/09 22:04:43,625 WARNING [139686479652608] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Controls white balance of color image. Setting any value will disable auto white balance
 17/09 22:04:43,625 WARNING [139686479652608] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Enable / disable auto-exposure
 17/09 22:04:43,625 WARNING [139686479652608] (sensor.cpp:632) Exception was thrown when inspecting Raw RGB Camera property Enable / disable auto-white-balance

RealSenseSupport, 2020-09-21:
@viisautta So do you mean that you can open the RealSenseViewer GUI(as below) after run ./realsense-viewer? But it crashed after start streaming? 
[image]

RealSenseSupport, 2020-09-27:
@viisautta Do you still have issue? Could you please update? Thanks!

RealSenseSupport, 2020-10-10:
@viisautta Did you get it through? Looking forward to your update. Thanks!

viisautta, 2020-10-11:
@RealSenseSupport
Oh, sorry not to update.
Now I am using 2.39.0 for some days. Such problem don't appear once more.
Thank you.

viisautta, 2020-10-11:
sorry again.

closed 2020-10-11 by viisautta