realsenseai/librealsense #7233: Intel RealSense L515 sensor not responding in RealSense Viewer v2.38.1

AlaiMich, 2020-08-31:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|

Intel RealSense L515 LiDAR sensor unstable and mostly doesn't show as connected even though it is. I just got the device and installed all the recommended apps per the manufacturer. Sometimes I have to reboot my machine in order for the device to work but it's not consistent. Is there any way to determine if the sensor is faulty? I'm also pairing the sensor with DOT3D app and haven't had any stable sessions. Device keeps disconnecting for unknown reasons. Any help is appreciated.

dorodnic, 2020-08-31:
Hi @AlaiMich 
Please check in the device manager if any of the relate devices have exclamation mark. This can be an indication of the device being connected as USB2 (due to cable or USB controller). This can explain the device not showing up in the Viewer.

RealSenseSupport, 2020-09-15:
Hi @AlaiMich 

Were you able to check the items that dorodnic outlined?  Having some understanding of the system seeing the device as USB3 or USB2 can have an effect on whether or not the device is detected properly.

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-22 by RealSenseSupport