realsenseai/librealsense #7235: L515 "No Frame Received"

cja-515, 2020-08-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.5.0.0 | 
| Operating System & Version |  Win 10 Pro  -18362.1016 | 
| Platform                                 | PC|
| SDK Version                            |   Intel RealSense  SDK 2.0 (v2.38.1/v2.36.1)                     | 

### Issue Description
Hi, I used  Intel RealSense  SDK 2.0 in win10 pro to test the stability of the depth sensor in the past two days. When it worked in the environment of 28℃~35℃, it  displayed "No Frame Received".I I had upgraded the firmware to the latest version, but this error is still displayed.  Is there any solution?
[image]

RealSenseSupport, 2020-09-15:
Hi @cja-515 

Is there anyway you can provide some further information about what you're seeing?  Is it specific to the temperature range that you outlined?  So if the temp is below that range, does the device function properly?  

In what setup or environment are you successful in using the L515?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-22 by RealSenseSupport

nh-97, 2021-03-01:
I have a problem using the L515 camera. When the intel Realsense viewer is turned on the depth, IR, confidence streams in the depth sensor will fail (No frame received!) for 1-2 seconds, then it works normally. But with RGB sensor, it works well
[image]
I have yet to find a way to resolve this issue.
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 
| Firmware Version                   | (Open RealSense Viewer  v2.38.1) | 
| Operating System & Version |   {Win (10)  | 
|USB cable                                | 3.2            |
| Platform                                 | PC |
| Language                            |  {C/C# }                          |