realsenseai/librealsense #7236: Is this typical noise for the L515?

SaxonDruce, 2020-08-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515 | 
| Firmware Version                   | 1.05.00.00 | 
| Operating System & Version | Windows 10 | 
| Viewer Version                            |  2.38.1                          | 

### Issue Description

Hi,

The datasheet for the L515 says that at 1m range (of a 95% reflectivity object), the accuracy is <5mm and the std deviation of the noise is 2.5mm. We'd expect 99.7% of the samples to be within 3 std deviations, and so (most of) the noise should be within +/- 7.5mm.

[image]

I've taken some test scans, and I'm wondering if the noise I'm seeing is typical. Or could there be something wrong with my device?

This is the scanner looking directly at a light coloured wall, from about 1m.

[image]

The scan has numerous ripples running through it. The ripples undulate constantly (like a waving ocean). Is that normal?

[image]

In this picture I have a couple of test objects on my floor. A red box about 13x9cm and about 3cm thick, and a white rock about 4x6cm and about 2cm thick.

[image]

This scan is from about 75cm above the floor. I'd expect noise (up to 7.5mm for 3 std devs) to be less than 1/4 the thickness of the box, or 1/3 the thickness of the rock. The ripples in this scan are generally not very large. The shape of the red box is fairly clear. However, the rock is quite distorted (passing below the floor). Could this be related to the curvature of the rock?

[image]

These are the same two objects on some dirt/gravel ground.

[image]

This is the corresponding scan, from about 75cm. Note that this ground was quite low in reflectivity, I had to get closer to it (under 1m) for it to be clearly visible in the scan. In this scan the rippling noise is quite extreme, and while the box is visible, the white rock is unrecognisable in the noise.

[image]

Does this look roughly like the expected behaviour of the scanner, or could something be going wrong here?

Thanks,
Saxon

blakely3dnm, 2020-08-31:
I too am curious about the noise under ideal conditions. I will let others speak to that. One issue I noticed is that your last image looks like it is in ambient sun light. In my hands *any* sunlight will increase noise and consequently decrease range. I have resorted to using VGA mode and really minimizing any sunlight in the scene.

SaxonDruce, 2020-09-01:
I just thought I'd include a gif of the noise on the flat wall (the first scan above), to better show what I mean:

[image]

Yes @blakely3dnm the last photo was taken outside in sunlight. Although, it was a cloudy overcast day (it was about to rain) and also later in the afternoon, so it wasn't direct sunlight just ambient light. Overall it was probably about as light as a reasonably well lit indoor space. Have you seen even ambient light outside cause problems, as opposed to direct sunlight?

Thanks.

SaxonDruce, 2020-09-01:
Here also is a gif of the noise in the indoor scene on the floor. In this scan the objects are about 85cm away. The scan is on an angle - the near part of the floor is at about 60cm, the far part at about 110cm.

[image]

blakely3dnm, 2020-09-01:
What is your sensor gain for the short range tests you are doing? Check this post for more ideas:
https://www.intelrealsense.com/optimizing-the-lidar-camera-l515-range/

And yes, in my experience ambient light through clouds will have enough IR light to cause problems. It is not the visible light that causes problems, so you could use it in a well-lit room just fine as long as there are no IR sources.

RealSenseSupport, 2020-09-17:
Also including direct link to the User Guide:  https://www.intelrealsense.com/download/13061/

Tuning the device for various situations may result in better images as outlined in the optimizing page as well as the L515 user guide.  Ambient light also will influence the resulting images as well.

RealSenseSupport, 2020-09-30:
Is there anything more needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

SaxonDruce, 2020-10-01:
I had been meaning to come back to post that I had done some additional tests at night time, and the noise was much better.

This was my test scene:

[image]

My test objects were:

- red box about 13x9cm and about 3cm thick
- white rock about 6x4cm and about 2cm thick
- grey rock about 6x5cm and about 2cm thick
- brown rock about 7x6cm and about 4cm high at the peak

The L515's camera didn't really pick up much colour (my phone does a better job in the photo above), but the depth is pretty good with much less noise than my earlier scan during day time:

[image]

Also for reference here it is coloured by depth:

[image]

These scans were made using the default settings for sensor gain.

My previous indoor test scans were done at daytime in a room a few metres from a large window, so I think this was also the cause of some of the added noise for the indoor scene.

Thanks @blakely3dnm and @RealSenseSupport for the info about avoiding sunlight as much as possible.

RealSenseSupport, 2020-10-07:
Glad you're able to get better depth information and data.  Thanks for following up and letting us know.

closed 2020-10-07 by RealSenseSupport