realsenseai/librealsense #7237: L515 Python API

salimifarzad, 2020-08-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { L515 } | 

| Operating System & Version |   {Linux(Ubuntu)| 
| Kernel Version (Linux Only)    |  (18.04)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {Python }                          | 
| Segment			|  {Robot }                          | 

### Issue Description
Hello,
I am newbie and using a L515 Lidar intel realsense and intend to use opencv as a tool in my python programming but I cant bring up the camera.
I aim to bring up the camera and capturing the images or recording video.
I would appreciate if you help me with a sample code. 
Best

dorodnic, 2020-08-31:
Hi @salimifarzad 
Please check you have python 2.7, 3.6 or 3.7 and install pyrealsense2 using `pip install pyrealsense2`. We provide pip packages for x86 architecture, if you are using Arm you may need to build from source. 
[wrappers/python](https://github.com/IntelRealSense/librealsense/tree/master/wrappers/python) offers some useful information. 
I also recommend following [this tutorial](https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb)

salimifarzad, 2020-09-02:
Thanks for your comment

closed 2020-09-03 by salimifarzad