realsenseai/librealsense #7240: USB cable for L515 camera Lidar

Vgexplore, 2020-08-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | L515
| Firmware Version                   ) | 
| Operating System & Version |   {Win 10  | 
| Kernel Version (Linux Only)    |                                        | 
| Platform                                 | PC|
                     | 

### Issue Description

We purchased the LIDAR L515 camera and we intend to use it to inspect underground cavities, passing via a borehole. Which type of USB cable would be the best to collect data at the surface? (maximum camera depth 30-35 m) ?

thupalo, 2020-08-31:
Hi, check DELOCK 83739, they have 20 and 50 meters, but $$$$

Vgexplore, 2020-09-02:
Thanks a lot Tadeusz !

RealSenseSupport, 2020-09-17:
Is there anything else needed on this thread?

*If we don’t hear from you in 7 days, this issue will be closed.

closed 2020-09-30 by RealSenseSupport