realsenseai/librealsense #7242: RGB Exposure causing major drop in frame rate

FLS-BP-US, 2020-08-31:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435i, L515 } | 
| Firmware Version                   | (01.05.00.00) | 
| Operating System & Version |   {Win (8.1/10) |
| Platform                                 | PC  |
| SDK Version                            |  {2.38.1}                          | 

### Issue Description
When I adjust the exposure of the RGB stream the frame rate drops in normally lit room.  Below are values collected with the L515.
312 == 30 fps
625 == 15 fps
1250 == 8 fps (Exposure matches reality).
2500 == 4 fps
I have observed this with the 435i.

If I set the deprioritize the auto-exposure option the frame rate stays at 30 fps.

RealSenseCustomerSupport, 2020-09-04:
----------------------------------------------

Hi @FLS-BP-US

When you disable auto-exposure that will make L515/D415i to adopt the exposure time you set
then fps will changed accordingly even you set the camera fps as 30fps,
The larger exposure time, the lower fps since the sensor take longer exposure time before output each frame.

The auto-exposure option should only work when auto-exposure is on
sometimes the AE algorithm decide to use larger exposure time (in dark environment) that is similar to you set a big exposure time in manual mode (e.g 1250)
if you want to keep the 30fps you set, you can uncheck the auto exposure priority option.

FLS-BP-US, 2020-09-04:
Thank you, I should have realized that from the numbers.

closed 2020-09-04 by FLS-BP-US