realsenseai/librealsense #7243: Hand tracking in librealsense 2.0

naihongtang, 2020-08-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |D435                                                        | 
| Firmware Version                   | 05.12.06.00                                             |
| Operating System & Version |  Win10                                                    |
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC                                                          |
| SDK Version                            |  {  2.36 }                                                 | 
| Language                               |  {C/C#/python }                                      | 
| Segment			         |  {Robot/Smartphone/VR/AR/others }    | 

### Issue Description
From previous posts, Intel recommends use 3rd party, like Nuitrack to do hand tracking, but Nuitrack only support gesture recognition, no finger joints tracking support. From https://www.intelrealsense.com/hand-tracking-overview/, Intel plan to have finger tracking in new release. Is there any progress on this subject? Thanks!

MartyG-RealSense, 2020-09-01:
Hi @ntang996   I am not aware of any plans for Intel to officially add finger joint tracking as a built-in feature of RealSense SDK 2.0.  You can use RealSense 400 Series cameras to train a data-set to recognize joints.  Intel published a seminar about this subject on YouTube.

https://www.youtube.com/watch?v=VSHDyUXSNqY

ntang996, 2020-09-01:
Thanks, MartyG

closed 2020-09-01 by ntang996