realsenseai/librealsense #7244: RealSense SR305's time duration to turn on camera keeps getting longer when keeps turning it on/off

nguyen-khang-w, 2020-09-01:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {SR305 } | 
| Firmware Version                   | (01.05.00.00) | 
| Operating System & Version |   Android 7 | 
| SDK Version                            |{2.38.1}  | 
| Language                            | Java     | 

### Issue Description

I am using RealSense camera on Android and I'm encountering a problem which is that the time duration to turn on the camera will keep getting longer as I keep turning it on/off

MartyG-RealSense, 2020-09-01:
Hi @nguyen-khang-w  The only lag-related Android wrapper issue that I currently know of is one where there can be lag in the camera preview when both color and depth streams are active.  

Does that describe your problem too please, or do you feel that it is different?

https://github.com/IntelRealSense/librealsense/issues/6817

MartyG-RealSense, 2020-09-09:
Hi @nguyen-khang-w   Do you still require assistance with this case, please?   Thanks!

nguyen-khang-w, 2020-09-12:
Thank you, I have solved this problem, when I destroy the Pipeline when the camera is turned off and the Config is set to static

MartyG-RealSense, 2020-09-12:
Great news - thanks so much for the update!   I will close this case now that you have found a solution.  Thanks again.

closed 2020-09-12 by MartyG-RealSense