realsenseai/librealsense #7245: get_distance() reading zero distance only

DavidCadtec, 2020-09-01:
I recently recorded (with a D415 camera...)  a bag file for some tests that i needed to run to track pedestrians... the issue is that  i simply cannot extract any value different than zero (0.00) with: 

distanceMeasure = depth_frame.get_distance(xC, yC)

Where  depth_frame is exactly the image extracted with:
__config.enable_device_from_file("/home/david/PycharmProjects/PyRealSense_Estudos/PeopleTrackTst2.bag")
frames = pipeline.wait_for_frames()
pipe_profile = pipeline.start(config)
depth_frame = frames.get_depth_frame()_

And so forth...
xC and yC are the coordinates (in Pixels) of the center of the ROI in the image. 

The curious thing is that applying Colorizer over the depth frames results in the (correct) displaying of the Jet Color gradients ... via:
colorizer = rs.colorizer(0)
depth_color_frame = colorizer.colorize(depth_frame)_

Again, this colorizer results with the proper displaying of jet the colors, confirming that the depth data is there ! Otherwise the colorizer would not display the depth color gradients... right ?

Is there anything i should be aware of when using  _depth_frame.get_distance(x,y) applied over frames read from a bag file ? Should i apply this to an average of the distance captured on an array of the x,y instead ? 
Should try to configure the stream and force the usage of the depth data inside the stream ?

**Running with the Python Wrapper over a bag file captured  from a  D415 on an  Ubuntu 16.4**

MartyG-RealSense, 2020-09-01:
Hi @DavidCadtec   Are you performing depth to color alignment on the bag data, like in Intel's Python **distance_to_object** tutorial that provides a distance value for a detected object (a dog in that particular tutorial).

https://github.com/IntelRealSense/librealsense/blob/jupyter/notebooks/distance_to_object.ipynb

DavidCadtec, 2020-09-04:
Hi @MartyG-RealSense 
Thank you four comments and suggestions !

Made some tests here .... and in fact, it seems that it's something related to the bag file i recorded...  

If i try  if i try to align Depth with Color frames... with my bag file it result in total black frames...
Other hand, if i try to use my code with the bag of the dog (used in the distance to Object example...), get_distance() will result the  correct values.. 

What makes me confused is that if i use depth_color_frame = colorizer.colorize(depth_frame) it displays the depth color gradient just as it shows the one in Distance_to_object example... so, the depth data is in fact in there right ?  
Will the result returned by: 
_depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()
depth = depth * depth_scale_

Be somehow different  different then the one resulted by
depth_frame.get_distance() method ?

MartyG-RealSense, 2020-09-04:
I would think that if the dog bag worked with the code then your own bag ought to work too.  

Are you able to replay your own bag file without problems if you drag and drop the file into the central panel of the **RealSense Viewer** program?  That will enable you to confirm that both the depth and color data is present in the file, and that playback of the file works.

I believe that if you are using the line below:

**depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()**

then it is looking for the first attached RealSense camera device that it can find, and is not looking at the bag file.  So any results generated from that command would be related to the attached camera and not the bag contents.

DavidCadtec, 2020-09-05:
Hi @MartyG-RealSense 

Yes, i can replay the bag file in RealSense Viewer program... both color and depth plays perfectly...  
No, in my code i'm not using: depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()
Just trying to read the depth in the depth_frame using:
depth_distance = depth_frame.get_distance(x,y)

Where:   depth_frame = frames.get_depth_frame()
x , y are the pixel location of the object detected in the image...

DavidCadtec, 2020-09-05:
Hi @MartyG-RealSense just an added issue:

I tried to include:
depth_scale = profile.get_device().first_depth_sensor().get_depth_scale()

And i got the error:
"Runtime error: Could not find requested sensor type !"

Is there any other mean to get the depth_scale parameter from a bag file ?

MartyG-RealSense, 2020-09-05:
Hi @DavidCadtec   There was a past case of someone who was reading a bag with pyrealsense2 and got that same RuntimeError error message when trying to read depth from it using the same method that you used, though they could read the bag without problems in the Viewer.   They found that updating pyrealsense2 fixed that particular problem.

https://github.com/IntelRealSense/librealsense/issues/4990

[image]

DavidCadtec, 2020-09-06:
Thanks @MartyG-RealSense 

Just to confirm... you mean upgrading like "sudo apt-get upgrade" ?
Or should i unistall the packages first... sorry for asking such basics ... but took so much effort and time to have PyRealSense working together  with many other packages i use in my Ubuntu that i prefer to ask first ...  i created a virtual environment in which i have all the packages installed...

Thanks !

MartyG-RealSense, 2020-09-06:
I have not heard of a means to "update" an existing pyrealsense2 installation, so I take it to mean removing and re-building pyrealsense2 to replace the currently installed version.  As you suggest, this would carry the risk of breaking a pyrealsense2 installation that is otherwise working.

MartyG-RealSense, 2020-09-18:
Hi @DavidCadtec   Do you require further assistance with this case, please?  Thanks!

DavidCadtec, 2020-09-23:
Yes... i do... so far i could not go any further in such a basic task like reading the depth data in a bag file, which, by the way, shows color and depth frames perfectly when played back in the RS Viewer. I simply don;t have a clue where to go from here.

MartyG-RealSense, 2020-09-23:
You are the David that I gave advice to about this subject on the other RealSense forum, aren't you? 

https://support.intelrealsense.com/hc/en-us/community/posts/360049966994

MartyG-RealSense, 2020-10-12:
Case closed due to no further comments received.

DavidCadtec, 2020-10-13:
Hi @MartyG-RealSense 

Pls accept my apologies for  taking so long to post this reply here ... had some personal issues and only now i am resuming this research.

Your suggestion for using the method get_data() of the depth_frame object worked fine !

As you recommended i extracted the 16 bit raw integer and then multiplied by 0.001 and the result has become under the real distance tolerance ...

Thank you, for you recommendations !

Now i'm looking for using the Intrinsics and extrinsics of the camera in order to get the aproximated position of the moving object inside the location in global coordinates.

Again, thank you for your guidance !

MartyG-RealSense, 2020-10-13:
There is a tutorial that uses

closed 2020-10-13 by MartyG-RealSense