realsenseai/librealsense #7247: D455 + Android Connection Issues

rafaelspring, 2020-09-01:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D455 | 
| Firmware Version                   | 05.12.06.00 | 
| Operating System & Version |  Android 10  | 
| SDK Version                            |  2.36.0  | 
| Segment			|  Tablets / Phones | 

### Issue Description
The D455 is not recognized as a USB device on Android. We tested this on a Samsung Galaxy Tab S6 (which has a USB 3.1 compliant USB-C port).

The tablet seems to completely ignore the D455. No Android / system logs indicate its presence. For example when connecting a D415 we immediately get system logs indicating the presence of a new USB device. None of that happens with the D455. It's as if no device was connected to USB.

We have not tested this on any other Android devices so it may be an issue specific to the S6 tablet.

RealSenseSupport, 2020-09-17:
@rafaelspring We tested on Android 8.1 RK3399. D455 can be recognized and start streaming successfully. What's the output of "lsusb |grep 8086" in adb shell of your setup when you get the issue? Looking forward to your update. Thanks!

RealSenseSupport, 2020-09-24:
@radfordi Did you get chance to try "lsusb |grep 8086" in adb shell? Looking forward to your reply. Thanks!

RealSenseSupport, 2020-10-10:
@radfordi Any update from your side? Thanks!

RealSenseSupport, 2020-10-21:
@radfordi We've got your NDA ticket and resolved this. Close it accordingly. Thanks!

closed 2020-10-21 by RealSenseSupport