realsenseai/librealsense #7250: Intel RealSense Viewer / Depth Quality Tool has crashed

cmello418, 2020-09-02:
| | |
|---|---|
|**librealsense**|2.38.1 RELEASE|
|**OS**|Windows|

Intel RealSense Viewer / Depth Quality Tool has crashed with the following error message:
```
Program terminated due to an unrecoverable SEH exception:
Illegal CPU instruction!
Possibly newer CPU architecture is required
```

I'm running on Windows 10, 64-bit  Core i7 2600K CPU

MartyG-RealSense, 2020-09-02:
Hi @cmello418   That error can be generated if you are using the Viewer with a CPU that is older than 6th Generation (Skylake).   Your CPU is an old 2nd Generation (Sandy Bridge).  This limitation only applies to the version of the Viewer that is installed by the Windows SDK installer from the Releases page and is due to a reliance by that Viewer version on the **OpenVINO** software.

If you download the 32.5 MB standalone version of the Viewer **(Intel.RealSense.Viewer.exe**) from the Assets list on the Releases page then this version can be used with any age of Intel processor.  Please note that this version will need to be launched from the folder that you downloaded it to instead of the desktop launch icon, unless you create a new shortcut icon for the desktop for the standalone version.

dorodnic, 2020-09-06:
@cmello418 - the answer from @MartyG-RealSense is accurate. In particular, OpenVINO is complaining about lack of AVX2 support in this specific CPU model.

MartyG-RealSense, 2020-09-18:
Case closed due to solution provided and no further comments received.

closed 2020-09-18 by MartyG-RealSense